meshtastic/firmware #1739: [Bug]: RAK4631 19003 boot loop

tunip, 2022-10-02:
### Category

Other

### Hardware

Rak4631

### Firmware Version

1.3.41.80ddb81

### Description

After flashing 1.3.41 my second RAK4631 is in a boot loop.

My first RAK4631 looks like this:
//\ E S H T /\ S T / C

??:??:?? 0 Filesystem files:
??:??:?? 0   RAK (88 Bytes)
??:??:?? 0   MODE (1 Bytes)
??:??:?? 1   db.proto (147 Bytes)
??:??:?? 1   config.proto (90 Bytes)
??:??:?? 1   module.proto (24 Bytes)
??:??:?? 1   channels.proto (53 Bytes)
...

But my second one reboots after `RAK (88 Bytes)` message.

Flashing the RAK4631 with a Arduino Example Sketch it works fine.

### Relevant log output

```Shell
//\ E S H T /\ S T / C

??:??:?? 1 Filesystem files:
??:??:?? 1   RAK (88 Bytes)
??:??:?? 1    (723605420 Bytes)⸮⸮ @??:??:?? 1 

//\ E S H T /\ S T / C

??:??:?? 1 Filesystem files:
??:??:?? 1   RAK (88 Bytes)
??:??:?? 1    (723605420 Bytes)⸮⸮ @??:??:?? 1 
...loop...

After reflashing bootloader & firmware:
//\ E S H T /\ S T / C

??:??:?? 1 Filesystem files:
??:??:?? 1   RAK (88 Bytes)
??:??:?? 1    (186734508 Bytes)⸮⸮ @??:??:?? 1 

//\ E S H T /\ S T / C

??:??:?? 1 Filesystem files:
??:??:?? 1   RAK (88 Bytes)
??:??:?? 1    (186734508 Bytes)⸮⸮ @??:??:?? 1 
...loop...
```

garthvh, 2022-10-02:
There is a reset .uf2 file that is pinned to the #nrf52 channel on discord, try dropping that on to the drive and then connecting to it via serial (meshtastic --noproto)  That will wipe all saved data from the device.

tunip, 2022-10-02:
Many thanks, that helped! Device works again. 👍

closed 2022-10-02 by tunip