meshtastic/firmware #1740: 1.3 Remaining Position Flag Issues

garthvh, 2022-10-02:
SeqNo, Speed and Heading are not working when their flags are enabled.  The latest iOS 16 app release allows for download of all the fields on the position log.  You can set the position flags under settings > position

caveman99, 2022-10-03:
some of the values were not computed, or passed. can you try https://github.com/meshtastic/Meshtastic-device/actions/runs/3176531657 ?

garthvh, 2022-10-03:
SeqNo is working perfectly.

Speed has a number now, but my tbeam is not moving and the value has been between 80 and 600 (meters per second?)

Heading keeps coming in as 0, I will test on a drive later.

garthvh, 2022-10-03:
Heading came in with values

caveman99, 2022-10-04:
i updated the speed calculation, i copy/pasted wrong stuff. the value from the lib is already in km/h ...

caveman99, 2022-10-04:
Heading should only update while moving faster than 'by foot' - even the comments in the code say this is for vehicle operation only. I have some compass modules with I2C here i intend to add, not for telemetry but for (optional) heading while stationary...

caveman99, 2022-10-04:
Please reopen if not satisfied :-)

garthvh, 2022-10-04:
It all seems to be working, not sure the format for the heading but it periodically gets data, speed and seqno seem good now

garthvh, 2022-10-04:
Speed and heading seem good, seqNo goes 1-8 and then starts over again

garthvh, 2022-10-05:
After testing more this seems fine

closed 2022-10-05 by garthvh