meshtastic/firmware #1761: [Feature request] Add ability to delete/remove individual nodes from nodedb

tropho23, 2022-10-06:
Request ability to delete/remove individual nodes from nodedb, rather than clearing the entire database. It is useful to keep some historical nodes in the database, such as those for lost devices, old devices that may come online again (such as fixed nodes with dead batteries).

Since `--set reset-nodedb` is currently used for clearing nodedb, recommend syntax `--set remove-nodedb <node name>` or similar.

I do not recommend allowing manually _adding_ nodes as that could be considered tainting or 'poisoning' the node database with false or incorrect node information.

tropho23, 2022-10-06:
I forgot to add that this should allow app developers to call this possibly admin message, to add actions for removing nodes withing the iOS and Android apps.

caveman99, 2022-10-09:
There's a recent addition to throw out the oldst node in the db if it is full. Would that be a good low key alternative?

tropho23, 2022-10-09:
No, but this feature request is an extension of that exact feature.  Just looking for a means to remove stale nodedb info selectively without deleting all nodedb info, if that is possible.  The current reset-nodedb feature works great...from a wipe-it-all perspective :)

tavdog, 2023-10-07:
So the first step here would be to implement this is firmware, triggerable via cli, then add trigger to the UI of android and ios.  A function in the firmware would take the node id number and delete it from memory and delete it from flash.

I'm looking through the code now.

garthvh, 2023-10-07:
You would create an admin message in the protobufs that can be consumed by any client

tavdog, 2023-10-08:
> You would create an admin message in the protobufs that can be consumed by any client

Yes now I see it would require modifying the adminMessage protobuf.  Is it frowned upon to go modifying the protobuf definitions for small features such as this ?

closed 2023-11-16 by thebentern

tropho23, 2023-11-16:
Excellent!  I'm happy to see this closed and completed 🙂

Tony

Closed #1761<https://github.com/meshtastic/firmware/issues/1761> as completed via #2941<https://github.com/meshtastic/firmware/pull/2941>.

PR #2941 (2023-11-16): Added Remove node by nodenum admin message
Closes #1761