meshtastic/firmware #1779: Feature suggestion: Support for serial receipt printers (module?)

sgofferj, 2022-10-09:
Cheap serial POS/receipt printers can be bought from sources like AliExpress for around $20. They are available with a TTL serial port and basically print what they receive via serial. With a few extensions, they can also print different fonts, weights and even graphics. Those extensions are defined in the ESC/POS standard and an Arduino library is available here: https://github.com/gdsports/ESC_POS_Printer and here: https://github.com/adafruit/Adafruit-Thermal-Printer-Library

Although it would be possible to e.g. print a QR code, I would limit my proposal to printing of received text messages. The printout would include the channel, the sender MAC and callsign as welll as the target MAC and callsign and the message, of course.

garthvh, 2022-10-09:
I think the right approach to this is to complete this issue https://github.com/meshtastic/Meshtastic-device/issues/1204 and then do your interaction with the serial printer from a second serially connected MCU

sgofferj, 2022-10-10:
@garthvh 
I'm  not sure about that. I forgot in my original post to mention that the printer would be connected to a second serial port on the Meshtastic module.
Using the "main" serial port through a second MCU adds a lot of complexity. The middleware would have to be maintained with regards to the protobufs. Also, using the "main" serial port would "lock" the Meshtastic module to the printer. Every time, something else would want to connect via the "main" serial port, say, a PC to update the firmware or to also receive messages, the printer would have to be disconnected.

If the module itself would just print out the messages through an additional port, the printer could stay connected permanently while the module would still be useable via USB for PCs/servers/RPIs/...

sgofferj, 2022-10-10:
I have like 5 or so different models in various projects and unless you want to print barcodes or different font sizes, they litterally print what is sent to them in US ASCII code - no tinkering required. Only baud rate varies between 2400, 9600 and 19200. The only thing needed would be to filter out unicode stuff like emojis or so.

garthvh, 2022-10-16:
Seems like a perfect serial module mode.

garthvh, 2022-11-20:
The now working txtmsg or nmea (position) serial modes should work great for testing this.

closed 2022-11-20 by garthvh