meshtastic/firmware #1781: [Bug]: Error:we don't yet have our myNodeinfo

lewisxhe, 2022-10-10:
### Category

Other

### Hardware

T-Beam

### Firmware Version

1.3.42

### Description

Use the following command to import version 1.3.42

```

.\device-install.bat -p COM28 -f firmware-tbeam-1.3.42.9bd9252.bin

```

After the t-beam v1.1 is swiped in, open the app. After successfully pairing Bluetooth, it returns when sending a message in the sending message

```
Error:we don't yet have our myNodeinfo
```

I use cli to obtain the following information:

```

> meshtastic.exe --port COM28  --info
Connected to radio

Owner: Meshtastic 3030 (3030)
My info: { "myNodeNum": 4066783280, "maxChannels": 8, "firmwareVersion": "1.3.42.9bd9252-d", "rebootCount": 7, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "hasWifi": true }
Nodes in mesh:  {'num': 4066783280, 'user': {'id': '!f2663030', 'longName': 'Meshtastic 3030', 'shortName': '3030', 'macaddr': '08:3a:f2:66:30:30', 'hwModel': 'TBEAM'}, 'position': {}}

Preferences: { "device": { "serialEnabled": true }, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "region": "EU_868", "hopLimit": 3, "txEnabled": true, "txPower": 27 }, "bluetooth": { "enabled": true, "fixedPin": 123456 } }

Module preferences: { "mqtt": {}, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/e/#CgMSAQESCDgDQANIAVAb

---
```

I try to set the region manually, but the error is the same after setting!

```
> meshtastic.exe --port COM28  --set lora.region EU_868

> meshtastic.exe --port COM28  --get lora.region
Connected to radio

lora.region: 3
Completed getting preferences

```

### Relevant log output

```Shell
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[... 175 lines cut ...]
??:??:?? 22 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 32 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
??:??:?? 32 [NodeInfoModule] sending owner !f2663030/Meshtastic 3030/3030
??:??:?? 32 [NodeInfoModule] Update DB node 0xf2663030, rx_time=0
??:??:?? 32 [NodeInfoModule] handleReceived(LOCAL) (id=0x0df86d0c Fr0x30 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=4 WANTRESP priority=10)
??:??:?? 32 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
??:??:?? 32 [NodeInfoModule] Add packet record (id=0x0df86d0c Fr0x30 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=4 WANTRESP priority=10)
??:??:?? 32 [NodeInfoModule] Should encrypt MQTT?: 1
??:??:?? 32 [NodeInfoModule] Expanding short PSK #1
??:??:?? 32 [NodeInfoModule] Using AES128 key!
??:??:?? 32 [NodeInfoModule] ESP32 crypt fr=f2663030, num=df86d0c, numBytes=50!
??:??:?? 32 [NodeInfoModule] enqueuing for send (id=0x0df86d0c Fr0x30 To0xff, WantAck0, HopLim3 Ch0x8 encrypted priority=10)
??:??:?? 32 [NodeInfoModule] txGood=1,rxGood=0,rxBad=0
??:??:?? 32 [RadioIf] Starting low level send (id=0x0df86d0c Fr0x30 To0xff, WantAck0, HopLim3 Ch0x8 encrypted priority=10)
??:??:?? 32 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=66, time 1148 ms
??:??:?? 32 [RadioIf] AirTime - Packet transmitted : 1148ms
??:??:?? 33 [RadioIf] Completed sending (id=0x0df86d0c Fr0x30 To0xff, WantAck0, HopLim3 Ch0x8 encrypted priority=10)
??:??:?? 42 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 62 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 82 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 102 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 122 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 142 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 162 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
```

caveman99, 2022-10-10:
The part in your Log right after `Not using WIFI` looks garbled. Did you set non-ascii letters as the nodename (owner) either long or short?

lewisxhe, 2022-10-10:
I did not set the owner. Everything is brand new. Before writing the firmware, I ran erase. I changed the owner to dev1, which also failed to send messages
```
meshtastic. exe --port COM28  --set-owner "dev1"
Connected to radio
Setting device owner to dev1
```

The log is as follows

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
斆\0@??:??:?? 0 

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files:
??:??:?? 0   /prefs/channels.proto (53 Bytes)
??:??:?? 0   /prefs/config.proto (79 Bytes)
??:??:?? 0   /prefs/db.proto (139 Bytes)
??:??:?? 0   /prefs/module.proto (22 Bytes)
??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 axp192/axp2101 PMU found
??:??:?? 0 I2C device found at address 0x3c
??:??:?? 0 0x4 subtype probed in 2 tries 
??:??:?? 0 ssd1306 display found
??:??:?? 0 2 I2C devices found
??:??:?? 0 Meshtastic hwvendor=4, swver=1.3.41.80ddb81-d
??:??:?? 0 Setting random seed 3796418115
??:??:?? 0 Total heap: 239500
??:??:?? 0 Free heap: 207864
??:??:?? 0 Total PSRAM: 4194252
??:??:?? 0 Free PSRAM: 4194252
??:??:?? 0 NVS: UsedEntries 118, FreeEntries 512, AllEntries 630, NameSpaces 4
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 6
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 18
??:??:?? 0 Loading /prefs/config.proto
??:??:?? 0 Loaded saved config version 18
??:??:?? 0 Loading /prefs/module.proto
??:??:?? 0 Loaded saved moduleConfig version 18
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Loaded saved channelFile version 18
??:??:?? 0 Number of Device Reboots: 6
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 0, using UNSET
??:??:?? 0 region=0, NODENUM=0xf2663030, dbsize=1
??:??:?? 0 Saving /prefs/db.proto
??:??:?? 0 Saving /prefs/config.proto
??:??:?? 0 Saving /prefs/module.proto
??:??:?? 0 Saving /prefs/channels.proto
[I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
??:??:?? 1 Warning: Failed to find AXP2101 power management
[I][XPowersAXP2101.tpp:230] ~XPowersAXP2101(): ~XPowersAXP2101
[I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
??:??:?? 1 AXP192 PMU init succeeded, using AXP192 PMU
[D][XPowersAXP192.tpp:2010] setInterruptImpl(): setInterruptImpl DISABLE - 0xffffffffff
[... 112 lines cut ...]
??:??:?? 22 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 31 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
??:??:?? 31 [NodeInfoModule] sending owner !f2663030/dev1/dv1
??:??:?? 31 [NodeInfoModule] Update DB node 0xf2663030, rx_time=0
??:??:?? 31 [NodeInfoModule] handleReceived(LOCAL) (id=0x3259b909 Fr0x30 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=4 WANTRESP priority=10)
??:??:?? 31 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
??:??:?? 31 [NodeInfoModule] Add packet record (id=0x3259b909 Fr0x30 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=4 WANTRESP priority=10)
??:??:?? 31 [NodeInfoModule] Should encrypt MQTT?: 1
??:??:?? 31 [NodeInfoModule] Expanding short PSK #1
??:??:?? 31 [NodeInfoModule] Using AES128 key!
??:??:?? 31 [NodeInfoModule] ESP32 crypt fr=f2663030, num=3259b909, numBytes=38!
??:??:?? 31 [NodeInfoModule] enqueuing for send (id=0x3259b909 Fr0x30 To0xff, WantAck0, HopLim3 Ch0x8 encrypted priority=10)
??:??:?? 31 [NodeInfoModule] txGood=1,rxGood=0,rxBad=0
??:??:?? 32 [RadioIf] Starting low level send (id=0x3259b909 Fr0x30 To0xff, WantAck0, HopLim3 Ch0x8 encrypted priority=10)
??:??:?? 32 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=54, time 1009 ms
??:??:?? 32 [RadioIf] AirTime - Packet transmitted : 1009ms
??:??:?? 33 [RadioIf] Completed sending (id=0x3259b909 Fr0x30 To0xff, WantAck0, HopLim3 Ch0x8 encrypted priority=10)
??:??:?? 42 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 62 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 82 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 102 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 122 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
??:??:?? 142 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0

```

caveman99, 2022-10-10:
thanks for confirming, looking into it.

andrekir, 2022-10-10:
please confirm the version of the Android app you are using is 1.3.41+

lewisxhe, 2022-10-10:
Well, it works. The latest version on the release page is 1.2.67. In fact, it has been updated to 1.3.xx. Now you can see the correct device name. It is normal. Thank you

closed 2022-10-10 by lewisxhe

jamieduk, 2023-05-24:
im getting error we dont have node info please help im in the android app using lora esp32 i siais update fwi click ok but not sure if its working as thats all it does. have you got a link to the newest fw  i can do via android if pos?

surlyhacker, 2023-05-26:
> im getting error we dont have node info please help im in the android app using lora esp32 i siais update fwi click ok but not sure if its working as thats all it does. have you got a link to the newest fw i can do via android if pos?

It's possible that this error is caused by incorrect bluetooth pairing with the device, assuming you're using bluetooth.
You can have your phone "forget" the Meshtastic device, and then go to your phone's bluetooth settings and explicitly pair with the Meshtastic device from there (which in my case asks me to enter the pairing PIN shown on the device screen, and it would be "123456" if your device has no screen).  Then I have to turn bluetooth off to disconnect the newly paired device.  And after turning bluetooth on, finally the Android app can connect to it fully and retrieve its configuration, eliminating this error.

I think the settings tab in the app where you connect to a device has room for improvement in its design, and it is also buggy.  It also does not really provide clear status information as to what it is doing.

DooMMasteR, 2024-03-10:
For me it does not work at all either, I tried to pair my T-Echo multiple times and the pairing seemt to work just fine, but the connection does not work, at all.