meshtastic/firmware #1803: Feature request: NMEA 0183 Waypoint/position from nearby devices over serial

samuelhj, 2022-10-15:
Adding NMEA0183 $WPL output from serial port to send position locations of nearby nodes to a Garmin (or any other GPS device supporting $WPL strings via NMEA0183) to show Waypoints on said Garmin device of locations of nearby nodes. 
See this link for reference: https://gpsd.gitlab.io/gpsd/NMEA.html#_wpl_waypoint_location

This would be an extremely useful use case for Meshtastic to make it easier for ICE-SAR units to quickly assess locations of their own vehicles, as well as providing grounds for snowmobile rental companies in Iceland to monitor locations of client's snowmobiles in the event of a disaster or complete whiteout (which is extremely common on glaciers in the wintertime) and can lead to very long search and rescue operations instead of simply navigating straight to a known location of a tracker. (Link to an article of said snowmobile incident: https://www.yahoo.com/lifestyle/group-tourists-rescued-iceland-snowmobiling-195822369.html)

Currently we use a local program called Sitewatch which displays location sent over TETRA to a centralized server that outputs to a website. However getting this information straight to our GPS devices is incredibly beneficial as it removes the need for LTE and TETRA uplink which can be unreliable in the interior of Iceland. As well as to provide an easy method of using the GPS navigation to locate and find said devices.

garthvh, 2022-10-16:
@samuelhj can you take some pictures and post them here as you test this so we can get information added to the docs? CC @jfirwin

arduionoGP, 2022-11-09:
Another, possibly simpler to code option, would be to add the initial sending device's unique ID to the inside of the position payload and then use MQTT to communicate to a  central display/mapping location (pc/pi/tablet/cellphone). (see last example in MQTT section on how easy it is to set this sort of thing up). Current meshtastic problem with your use case is the ID of the initial sending device is non contained in the position packet and the device ID that is in the mqtt envelope is always the mqtt connected lora device.

garthvh, 2022-11-09:
All packets have a too and a from, what if are you missing?

arduionoGP, 2022-11-09:
It sounds like he wants to plot all of his devices on a map without using the android app.
The MQTT envelope contains {"sender": which in this case is set to the device connected to the mqtt broker - NOT the device originating the GPS location payload, "type:  which in this case would be GPS position, and "payload" which ends up being the coordinates but NOT the ID of the device generating those coordinates.} If the payload contained that, it could be plotted using MQTT without developing another protobuf and protocol for him.

garthvh, 2022-11-09:
It is already implemented in the serial module

samuelhj, 2022-11-09:
I've successfully managed to get this working and have tested it on my workbench, my hope is to field test this in the next few days. There is a known bug as of now that causes location to be off the charts. (longitude is reported in a negative value, which garmin doesn't like).

Settings for the NMEA from T-Beam to work with Garmin are these:

--set serial.txd 15 --set serial.rxd 35 --set serial.mode NMEA --set serial.baud BAUD_4800 --set serial.timeout 1000 --set serial.enabled true

Pin 15 is Tx *from* T-Beam to the RX on the Garmin device. This was tested with Garmin GPSMap276C.

A level converter is needed to switch from 3.3V of the T-Beam to the Garmin device. Something like MAX3323 or similar 3.3V capable level converter.

garthvh, 2022-11-10:
Awesome!

closed 2022-11-10 by garthvh