meshtastic/firmware #1811: [Bug]: fixed position is not preserved across reboots

IZ1IVA, 2022-10-16:
### Category

Other

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

1.3.44

### Description

When setting a fixed position, first I disable the GPS with `position.gps_enabled`, then reboot to make sure GPS stays off. I then set the position with `--setalt --setlat --setlon` and that works as expected.
However, a further reboot will set altitude, latitude and longitude to zero.
This is particularly inconvenient for remote nodes without GPS (e.g. T-Loras).

IZ1IVA, 2022-10-17:
> @IZ1IVA can you try updating to the the latest python cli version? I suspect this change might have resolved it: [meshtastic/Meshtastic-python#385](https://github.com/meshtastic/Meshtastic-python/pull/385)

I will! I'm sorry I've tried yesterday without first checking for an updated Python CLI...

caveman99, 2022-10-18:
I have a Router Node based on TLORA-2.1-1.6 running more or less 'Master' and it has he fixed position fine and keeps it across reboots and firmware updates.

```
config:
[...]
  position:
    fixedPosition: true
    gpsAttemptTime: 900
    gpsUpdateInterval: 120
    positionBroadcastSecs: 900
    positionFlags: 3
[...]
location:
  alt: 170
  lat: 50.<redacted>
  lon: 6.<redacted>
[...]
```

thebentern, 2022-10-18:
Seems like it's working correctly. Closing

caveman99, 2022-10-18:
please try a factory reset and then save the coordinates again.

IZ1IVA, 2022-10-18:
I confirm it does not retain a fixed position after a reboot, showing instead zeroes for altitude and coordinates.

thebentern, 2022-10-19:
Could you take some serial logs at startup via `meshtastic --noproto`? I'm wondering if there are some file system persistance on the device somehow.

IZ1IVA, 2022-10-19:
> Could you take some serial logs at startup via `meshtastic --noproto`? I'm wondering if there are some file system persistance on the device somehow.

Here you go, I've rebooted the board from the phone while capturing the log. Cheers!
[tlora boot dump.txt](https://github.com/meshtastic/Meshtastic-device/files/9818031/tlora.boot.dump.txt)

caveman99, 2022-10-19:
the position itself is not stored in the config.position proto, but in the nodeDB for your own nodenum.

caveman99, 2022-10-19:
'Nodes' Tab, index 0 should be your own node... 
[image]

ptrhess, 2022-10-19:
Seeing the same issue on a new Station G1 I just received today, with 1.3.45 firmware. Via the Python CLI, I set the position using `--setlat`, `--setlon` and `--setalt`, it shows up in the right place in the Nodes list, then I reboot and the information is gone. 

Before setting location:
```
Nodes in mesh:  {'num': 3446157492, 'user': {'id': '!cd6830b4', 'longName': 'RunValley', 'shortName': 'RnV', 'macaddr': 'e8:31:cd:68:30:b4', 'hwModel': 'STATION_G1'}, 'position': {}, 'lastHeard': 1666202430, 'deviceMetrics': {'airUtilTx': 1.1277223}}
```

After setting location but before reboot:
```
Nodes in mesh:  {'num': 3446157492, 'user': {'id': '!cd6830b4', 'longName': 'RunValley', 'shortName': 'RnV', 'macaddr': 'e8:31:cd:68:30:b4', 'hwModel': 'STATION_G1'}, 'position': {'latitudeI': 200000000, 'longitudeI': -600000000, 'altitude': 250, 'latitude': 20.0, 'longitude': -60.0}, 'lastHeard': 1666202548, 'deviceMetrics': {'channelUtilization': 47.948334, 'airUtilTx': 2.273639}}
```

After reboot:
```
Nodes in mesh:  {'num': 3446157492, 'user': {'id': '!cd6830b4', 'longName': 'RunValley', 'shortName': 'RnV', 'macaddr': 'e8:31:cd:68:30:b4', 'hwModel': 'STATION_G1'}, 'position': {}, 'deviceMetrics': {}}
```

Logs of these operations -- read info, set position, read info, reboot, read info: 
[station_g1_setpos_log.txt](https://github.com/meshtastic/Meshtastic-device/files/9823276/station_g1_setpos_log.txt)

thebentern, 2022-10-20:
@IZ1IVA @ptrhess I'm pretty confident I found and fixed the problem. Please try the build artifacts from the pull request below and confirm that it fixes the persistence.
https://github.com/meshtastic/Meshtastic-device/pull/1829

ptrhess, 2022-10-20:
@thebentern, tested with `1.3.46.5df4838-d` from #1829. Looks good. I ran it on the Station G1, device rebooted immediately after setting position, lat/lon/alt values were preserved post-reboot. I've rebooted a couple more times to verify. Thanks!

closed 2022-10-20 by thebentern

IZ1IVA, 2022-10-20:
> @IZ1IVA @ptrhess I'm pretty confident I found and fixed the problem. Please try the build artifacts from the pull request below and confirm that it fixes the persistence. #1829

I've only tested on a tlora so far, but it works! Will test it on the other boards next week.

PR #1829 (2022-10-20): Save devicestate (nodedb) on position config change - Fixes fixed position bug
Fixes #1811 (I think)
I believe what was happening is that only the position config changes were being saved to disk and then we trigger a reboot. So unless a NodeDB save for device state occured within that time, the position packet stored in the NodeDB wouldn't be persisted to the filesystem.