meshtastic/firmware #1819: [Bug]: when outgoing TX Queue is full messages are simple dropped without notification

pavelb-techspark, 2022-10-18:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

1.3.46

### Description

An attempt to send more than 16 messages discards some of them silently.

I propose to add a QueueStatus message to FromRadio that will notify host with the status of the TX Queue as well as latest attempt to queue a message.

garthvh, 2022-10-18:
Are you able to actually cause this with a connected client?

pavelb-techspark, 2022-10-18:
Yes, if you want to use Meshtastic as an inexpensive PHY layer, for instance for IP tunneling. Sending single packet can take up to 500 ms, so:

```
ping -i 0.3 $OTHER_HOST_IP
```

will easily cause this.

pavelb-techspark, 2022-10-18:
Additional information:
https://github.com/meshtastic/Meshtastic-python/pull/386#issuecomment-1283110084

garthvh, 2022-10-19:
It is a neat proof of concept, it may not be practical for a lora mesh. WLAN concepts break down some with flood routing and the bandwidth constraints of lora. Every node in the mesh will broadcast this tunnel and it seems like it will quickly exceed duty cycle limits.

ComBatVision, 2022-10-19:
Any transmitting device should have outgoing buffer control to be utilized properly.

Without buffer control it is only a toy to show school friends how cool to send some SMS via radio and be proud of it.

My opinion that without outgoing buffer control this device can not be used in any application where the stream of data exists, e.g. continuos telemetry from any sensors.

Not so many countries in the world have duty cycle limits. Think about all possible customers.

pavelb-techspark, 2022-10-19:
There are good reasons to ignore law limits in the life or death situations. And those who don't need this just wont use it.

We needed a reliable low-power low-bandwidth mesh, our packets being silently ignored was the missing part for us, and I believe it will be for some other users.

mc-hamster, 2022-10-19:
> There are good reasons to ignore law limits in the life or death situations. And those who don't need this just wont use it.
> 
> We needed a reliable low-power low-bandwidth mesh, our packets being silently ignored was the missing part for us, and I believe it will be for some other users.
> 
> edit: Flooding mesh might be reworked to a more dynamic-routing one with an appropriate module.

We purposely stopped working on a dynamic routing system. It introduces too much overhead for our limited capacities.

If you'd like to introduce one as an alternate configuration during initial setup, we'd entertain the option.

mc-hamster, 2022-10-19:
It's the DSR class in the 1.2 code.

closed 2023-01-04 by sachaw