meshtastic/firmware #1821: [Bug]: RAK11200 (ESP32-core) and RAK13300 dont except setting changes by CLI/web/app nor do send LoRa packets

djjayballz, 2022-10-18:
### Category

Hardware Compatibility

### Hardware

Rak11200

### Firmware Version

1.3.44

### Description

Hello,
i got some strange problems with my RAK11200 + RAK13300 setup. I'd like to use this combination for a LoRa<->internet-gateway but run into issues on first line.

I flashed the ESP-core with the device-install.sh-script provided by the 1.3.44-release (other older releases did not work too!).
`./device-install.sh -f firmware-rak11200-1.3.44.4fa8d02.bin`
This worked without a problem.

When I try to set the lora.region with `meshtastic --set lora.region EU_868` the device happily accepts this, but if I try to get the setting with `meshtastic --get lora.region` it outputs "region = 0", but it should be region = 3 for EU868. Even after rebooting the same behaviour exists.

But when i disconnect the LoRa-module (RAK13300) everything works fine. All commands for changing settings are accepted and stored in the ESP32. Even when using the webinterface in Chromium-browser changes can be made.
When i attach the RAK13300 and reboot the device, it does not change any settings, it always reconnects automatically in the webinterface and i can not send any LoRa message.

Interestingly i tried the [TX + RX examples for Arduino IDE provided by RAK wireless](https://github.com/RAKWireless/WisBlock/tree/master/examples/RAK11200/communications/LoRa). With that code I can send and receive packets.
So i guess the LoRa-module is not broken. There must be a problem with meshtastics code.

I attach the output of `meshtastic --noproto` after rebooting. (fresh flashed RAK11200 and RAK13300 attached)
The device reboots automatically.
I guess `assert failed: void SX126xInterface<T>::setStandby() [with T = SX1262] SX126xInterface.cpp:169 (err == RADIOLIB_ERR_NONE)` is the most interesting output provided.

Does anyone else have these problems?

Best wishes
jayballz

### Relevant log output

```Shell
meshtastic --noproto
WARNING file:mesh_interface.py _sendToRadio line:467 Not sending packet because protocol use is disabled by noProto
?Connected to radio
WARNING file:mesh_interface.py _sendPacket line:357 Not sending packet because protocol use is disabled by noProto
%=1%	}II}9=9?

Backtrace:0x40084131:0x3ffd67d00x40095f89:0x3ffd67f0 0x4009bc45:0x3ffd6810 0x400de631:0x3ffd6940 0x400de2f8:0x3ffd6960 0x400e1a32:0x3ffd6980 0x400e1a90:0x3ffd69a0 0x401f87ea:0x3ffd69c0 0x400d4f2d:0x3ffd69e0 0x400d4fde:0x3ffd6a00 0x400f59d5:0x3ffd6a20 0x400dd0a5:0x3ffd6a50 0x401227e1:0x3ffd6a70 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
WARNING file:mesh_interface.py _sendToRadio line:467 Not sending packet because protocol use is disabled by noProto
??:??:?? 0 

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files (471040/1048576 Bytes):
[... 286 lines cut ...]
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 0, using UNSET
[   386][E][vfs_api.cpp:104] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
??:??:?? 0 No /prefs/module.proto preferences found
??:??:?? 0 Installing default ModuleConfig
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Loaded saved channelFile version 19
??:??:?? 0 Number of Device Reboots: 5
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 0, using UNSET
??:??:?? 0 region=0, NODENUM=0x934c42e4, dbsize=1
??:??:?? 0 Saving /prefs/db.proto
??:??:?? 0 Read RTC time as 10
??:??:?? 0 Using MSL altitude model
??:??:?? 0 WANT GPS=1

^CTraceback (most recent call last):
  File "/home/jb/.local/bin/meshtastic", line 8, in <module>
    sys.exit(main())
  File "/home/jb/.local/lib/python3.8/site-packages/meshtastic/__main__.py", line 950, in main
    common()
  File "/home/jb/.local/lib/python3.8/site-packages/meshtastic/__main__.py", line 743, in common
    time.sleep(1000)
KeyboardInterrupt
```

caveman99, 2022-10-18:
please check the firmware from https://github.com/meshtastic/Meshtastic-device/suites/8810067067/artifacts/400865220 there was a recent bugfix on that RAK combo

caveman99, 2022-10-22:
by now this fix is included in the 1.3.46 firmware. Can you please try updating and rerun your test?

caveman99, 2022-10-26:
no feedback, assuming fixed

djjayballz, 2022-10-31:
Hello Caveman,

sorry, i had some stuff to do.
No unfortunately not. Nothing changed

`Connected Devices
2471248612

Ctrl+K
2471248612
2471248612
Meshtastic 42e4

Disconnect

Information
Battery State

0%

Unknown
Connected Peers

None Discovered.

Current Location

No Fix.

Config Status

Config

Module Config

Hardware

Nodes

Waypoints

Connection

Serial Logs
assert failed: void SX126xInterface<T>::setStandby() [with T = SX1262] SX126xInterface.cpp:169 (err == RADIOLIB_ERR_NONE)
Backtrace:0x40084131:0x3ffd68500x40095f89:0x3ffd6870 0x4009bc45:0x3ffd6890 0x400de885:0x3ffd69c0 0x400de54c:0x3ffd69e0 0x400e1d5a:0x3ffd6a00 0x400e1db8:0x3ffd6a20 0x401f941e:0x3ffd6a40 0x400d4f81:0x3ffd6a60 0x400d5032:0x3ffd6a80 0x400f5da5:0x3ffd6aa0 0x400dd245:0x3ffd6ad0 0x401233bd:0x3ffd6af0
ELF file SHA256: 0000000000000000
Rebooting...
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[ 11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
??:??:?? 0
//\ E S H T /\ S T / C
??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files (471040/1048576 Bytes):
??:??:?? 0 /prefs/channels.proto (53 Bytes)
??:??:?? 0 /prefs/db.proto (144 Bytes)
??:??:?? 0 /static/.gitkeep (0 Bytes)
??:??:?? 0 /static/Logo_Black.svg.gz (592 Bytes)
??:??:?? 0 /static/icon.svg.gz (842 Bytes)
??:??:?? 0 /static/index.2759ba5d.js.gz (347 Bytes)
??:??:?? 0 /static/index.494bd86d.css.gz (15113 Bytes)
??:??:?? 0 /static/index.fd944a00.js.gz (408815 Bytes)
[... 137 lines cut ...]
[ 1937][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
??:??:?? 1 Starting meshradio init...
??:??:?? 1 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
??:??:?? 1 Radio myRegion->freqStart / myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
??:??:?? 1 Radio myRegion->numChannels: 104
??:??:?? 1 Radio channel_num: 19
??:??:?? 1 Radio frequency: 906.875000
??:??:?? 1 Slot time: 42 msec
??:??:?? 1 Set radio: final power level=22
??:??:?? 2 SX126x init result 0
??:??:?? 2 Frequency set to 906.875000
??:??:?? 2 Bandwidth set to 250.000000
??:??:?? 2 Power output set to 22
??:??:?? 2 Current limit set to 140.000000
??:??:?? 2 Current limit set result 0
E (2018) gpio: gpio_set_level(226): GPIO output gpio_num error
E (2019) gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first
E (2027) gpio: gpio_set_level(226): GPIO output gpio_num error
??:??:?? 2 SX1262 Radio init succeeded, using SX1262 radio
??:??:?? 2 Not using WIFI
??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=237, time 3188 ms
??:??:?? 2 myNodeInfo.bitrate = 74.341286 bytes / sec
??:??:?? 2 PowerFSM init, USB power=1
??:??:?? 2 Ente��хѕ� BOOT
[ 2077][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz`

djjayballz, 2022-11-07:
little update with brand new RAK13300 Lora-Modules
`ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
WARNING file:mesh_interface.py _sendToRadio line:467 Not sending packet because protocol use is disabled by noProto
??:??:?? 0 

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files (475136/1048576 Bytes):
??:??:?? 0  /prefs/channels.proto (53 Bytes)
??:??:?? 0  /prefs/config.proto (85 Bytes)
??:??:?? 0  /prefs/db.proto (143 Bytes)
??:??:?? 0  /static/.gitkeep (0 Bytes)
??:??:?? 0  /static/Logo_Black.svg.gz (592 Bytes)
??:??:?? 0  /static/icon.svg.gz (842 Bytes)
??:??:?? 0  /static/index.2759ba5d.js.gz (347 Bytes)
??:??:?? 0  /static/index.85831067.css.gz (15140 Bytes)
??:??:?? 0  /static/index.d4fbecc1.js.gz (410906 Bytes)
??:??:?? 0  /static/index.html.gz (507 Bytes)
??:??:?? 0  /static/robots.txt.gz (42 Bytes)
??:??:?? 0  /static/site.webmanifest.gz (196 Bytes)
[   312][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=4 scl=5 freq=100000
??:??:?? 0 No I2C devices found
??:??:?? 0 Meshtastic hwvendor=13, swver=2.0.0.18ab874-d
??:??:?? 0 Setting random seed 4027512009
??:??:?? 0 Total heap: 201988
??:??:?? 0 Free heap: 168544
??:??:?? 0 Total PSRAM: 0
??:??:?? 0 Free PSRAM: 0
??:??:?? 0 NVS: UsedEntries 70, FreeEntries 560, AllEntries 630, NameSpaces 3
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 35
??:??:?? 0 OTA firmware version 0.1.1.98ab02e
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 20
??:??:?? 0 Loading /prefs/config.proto
??:??:?? 0 Loaded saved config version 20
[   434][E][vfs_api.cpp:104] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
??:??:?? 0 No /prefs/module.proto preferences found
??:??:?? 0 Installing default ModuleConfig
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Loaded saved channelFile version 20
[   486][E][vfs_api.cpp:104] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
??:??:?? 0 No /oem/oem.proto preferences found
??:??:?? 0 Number of Device Reboots: 35
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 3, using EU_868
??:??:?? 0 region=3, NODENUM=0x934c42e4, dbsize=1
??:??:?? 0 Saving /prefs/db.proto
??:??:?? 0 No I2C devices found
??:??:?? 0 Read RTC time as 21
??:??:?? 0 Using MSL altitude model
??:??:?? 0 WANT GPS=1
??:??:?? 1 Warning: Failed to find UBlox & MTK GNSS Module
??:??:?? 1 GxGSA NOT available
??:??:?? 1 External Notification Module Disabled
E (1970) gpio: gpio_set_level(226): GPIO output gpio_num error
[  1979][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
??:??:?? 1 Starting meshradio init...
[... 31 lines cut ...]
??:??:?? 11 [DeviceTelemetryModule] Node status update: 1 online, 1 total
??:??:?? 11 [DeviceTelemetryModule] Sending packet to mesh
??:??:?? 12 [DeviceTelemetryModule] Update DB node 0x934c42e4, rx_time=0
??:??:?? 12 [DeviceTelemetryModule] handleReceived(LOCAL) (id=0x0fa853d7 Fr0xe4 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=67 priority=1)
??:??:?? 12 [DeviceTelemetryModule] No modules interested in portnum=67, src=LOCAL
??:??:?? 12 [DeviceTelemetryModule] Add packet record (id=0x0fa853d7 Fr0xe4 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=67 priority=1)
??:??:?? 12 [DeviceTelemetryModule] Should encrypt MQTT?: 1
??:??:?? 12 [DeviceTelemetryModule] Expanding short PSK #1
??:??:?? 12 [DeviceTelemetryModule] Using AES128 key!
??:??:?? 12 [DeviceTelemetryModule] ESP32 crypt fr=934c42e4, num=fa853d7, numBytes=11!
??:??:?? 12 [DeviceTelemetryModule] enqueuing for send (id=0x0fa853d7 Fr0xe4 To0xff, WantAck0, HopLim3 Ch0x8 encrypted priority=1)
??:??:?? 12 [DeviceTelemetryModule] txGood=0,rxGood=0,rxBad=0
??:??:?? 12 [DeviceTelemetryModule] Using channel 0 (hash 0x8)
??:??:?? 12 [DeviceTelemetryModule] Expanding short PSK #1
??:??:?? 12 [DeviceTelemetryModule] Using AES128 key!
??:??:?? 12 [DeviceTelemetryModule] ESP32 crypt fr=934c42e4, num=fa853d7, numBytes=11!
??:??:?? 12 [DeviceTelemetryModule] decoded message (id=0x0fa853d7 Fr0xe4 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=67 priority=1)

assert failed: void SX126xInterface<T>::setStandby() [with T = SX1262] SX126xInterface.cpp:169 (err == RADIOLIB_ERR_NONE)

Backtrace:0x40084131:0x3ffd66e00x40095f89:0x3ffd6700 0x4009bc45:0x3ffd6720 0x400de565:0x3ffd6850 0x400de22c:0x3ffd6870 0x400e1a42:0x3ffd6890 0x400e1aa0:0x3ffd68b0 0x401f839e:0x3ffd68d0 0x400d4f51:0x3ffd68f0 0x400d5002:0x3ffd6910 0x400f572d:0x3ffd6930 0x400dcf15:0x3ffd6960 0x401223a5:0x3ffd6980 

ELF file SHA256: 0000000000000000

Rebooting...`

caveman99, 2022-11-15:
for starters 

```
E (1957) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1958) gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first
E (1966) gpio: gpio_set_level(226): GPIO output gpio_num error
```

most likely triggered by the -1 ... let's try setting it to RADIOLIB_NC

caveman99, 2022-11-15:
assert failed: void SX126xInterface<T>::setStandby() [with T = SX1262] SX126xInterface.cpp:173 (err == RADIOLIB_ERR_NONE)

Additional debugging reveals the assert eing triggered by an error -707 (RADIOLIB_ERR_SPI_CMD_FAILED) in SPIwriteCommand(RADIOLIB_SX126X_CMD_SET_STANDBY[...]

djjayballz, 2022-11-18:
Small update:

@caveman99 and i figured something out but we were not able to fix the bug.

By accident we found out that RAK11200 + RAK13300 on a RAK5005 baseboard work flawlessly, but if you change the baseboard to the new RAK19007 or the big RAK19001_VB the code crashes and core reboots.
--> something with the pinout?

Interestingly if you add a GPS-module (RAK1910) to the RAK19007 board (not tested yet on the RAK19001) the code works flawlessly again.
--> very strange behavior

closed 2023-02-11 by garthvh

TVio-Fish95, 2024-08-25:
@cavemann99 @djjayballz, what was the resolution here in the case of 19001? I'm using one and when attempting to flash, I go into DFU mode, enter 'meshastic --proto' into my cmd (in admin mode) and get the following errors: 

WARNING file:mesh_interface.py _sendToRadio line:880 Not sending packet because protocol use is disabled by noProto
Connected to radio
WARNING file:mesh_interface.py _sendPacket line:712 Not sending packet because protocol use is disabled by noProto

I get this result even after updating to latest meshtastic with 'pip install --upgrade meshtastic' command in the same terminal. 

Any advice? Newbie here... Thanks!