meshtastic/firmware #1833: [Feature Request]: Hook up PWM buzzers to the Bell Characters sent from Canned Messages and External Notifications

garthvh, 2022-10-21:
### Category

Other

### Hardware

Rak4631

### Firmware Version

1.3.46

### Description

Hook up the RAK buzzer to the external notification and canned messages module when bell characters are enabled.

caveman99, 2022-10-21:
the idea is to decide what to do on a definition basis? this would disable the possibility to set a custom output pin on devices with a PWM buzzer. The pin could do other things like switch on the lights...

garthvh, 2022-10-21:
Can we use pwm if available unless a flip value is set? The stuff that works via auto detect is a lot more likely to get used than flip configured stuff

caveman99, 2022-10-21:
yes, that would be an option :-)

lesykm, 2022-10-22:
Please definitely consider besides RAK-specific PWM buzzer also generic GPIO operation - it can be used to light alarm LED, regular (not PWM) buzzer like https://grobotronics.com/buzzer-5v.html - for this kind of devices only GPIO low/high is required (example https://www.instructables.com/Raspberry-Pi-Tutorial-How-to-Use-a-Buzzer/)
Thanks!

garthvh, 2022-10-22:
> Please definitely consider besides RAK-specific PWM buzzer also generic GPIO operation - it can be used to light alarm LED, regular (not PWM) buzzer like https://grobotronics.com/buzzer-5v.html - [...]

this works now, pwm is not rak specific

caveman99, 2022-10-22:
we assume a PWM buzzer is onboard when the hardware definition has PIN_BUZZER set. That's true at the moment for rak4631 and the m5stack (and two more obscure DIY boards). The m5stack even has its own PWM implementation in the variant folder.

I'll wire that in so that for these boards the PWM code is used if there's no notification pin set. If the pin is set we'll use that instead, like on the other boards.

caveman99, 2022-10-23:
this is live in firmware 1.3.47

https://github.com/meshtastic/Meshtastic-device/releases/tag/v1.3.47.05147c0

closed 2022-10-23 by caveman99