meshtastic/firmware #1835: Evaluate usefulness of SoftAP mode

garthvh, 2022-10-21:
The documentation has been updated to no longer advertise a bunch of SoftAP features that don't work at all.  We need to determine what exactly softap mode is useful for in the context of meshtastic, a lot of users assume this is an off the grid way they can serve the webui to a number of devices which has never worked in my testing.

Modern phones make connecting to a self signed cert / wifi with no internet problematic, and turning on SoftAP seems to really heat up the ESP32 chip.

@mc-hamster what is the expected SoftAP experience for users?

thebentern, 2022-10-21:
My experience with it is these ESP32 picos chips get unbelievably hot while hosting a SoftAP even worse than they do as a wifi client.

closed 2022-10-27 by garthvh

ulrichard, 2023-05-21:
Does that mean there is no SoftAP any more at all?
I would have a use case for it. My phones run ubports, PureOS or Mobian. Since there is no native meshtastic client for any of them, a website version would be great. Since it's all local, I'd be fine with http instead of https.

sachaw, 2023-05-21:
@ulrichard no need for it, make your phone the AP and have the device connect to it or use it over serial/Bluetooth with the web app

garthvh, 2023-05-22:
> Does that mean there is no SoftAP any more at all?
> I would have a use case for it. My phones run ubports, PureOS or Mobian. Since there is no native meshtastic client for any of them, a website version would be great. Since it's all local, I'd be fine with http instead of https.

Soft AP never provided a full webui anyways, it was a tool to save Wi-Fi credentials only.

ulrichard, 2023-05-22:
Indeed, thanks for the hint. 
Connecting with the phone AP works, bu there are issues:
* I can send messages, but the received messages are only displayed on the TBeam and not in the browser on the phone.
* The peers are not displayed on the map
* I can download map tiles when connected to the home Wifi, but when I switch to the AP mode, the map tiles are gone.
I read somewhere that the ubports morph browser is based on chromium, but serial and bluetooth are greyed out on the meshtastic connection dialog.

garthvh, 2023-05-23:
I don’t think chrome for android supports web serial