meshtastic/firmware #1836: Bug:  Unable to access Range Test CSV file if save is enabled

garthvh, 2022-10-21:


sachaw, 2022-10-22:
I've always struggled to even get the CSV file to be created

garthvh, 2022-10-22:
> I've always struggled to even get the CSV file to be created

It has been elusive

caveman99, 2022-10-22:
@sachaw the file browser in the current WebUI *should* work from the js perspective, right?

sachaw, 2022-10-22:
Yeah, no issues there

caveman99, 2022-10-22:
i got device to return something parseable and the WebUI displays it however there's something allocating and not freeing memory in the (new) json lib. After 10ish seconds/requests the ESP32 logs the mother of all core dumps and reboots.

caveman99, 2022-10-22:
https://github.com/meshtastic/Meshtastic-device/pull/1838 has the artefacts (soon). Sample coredump:

```
[HTTPS:I] Request: GET /json/fs/browse/static (FID=53)

abort() was called at PC 0x401bf0bb on core 1

Backtrace:0x40084145:0x3ffd68c00x40095f9d:0x3ffd68e0 0x4009bb55:0x3ffd6900 0x401bf0bb:0x3ffd6980 0x401bf102:0x3ffd69a0 0x401bf2dd:0x3ffd69c0 0x401bf398:0x3ffd69e0 0x400fc219:0x3ffd6a00 0x400e9f56:0x3ffd6a30 0x401fcd77:0x3ffd6b40 0x401034b6:0x3ffd6b60 0x40103ce3:0x3ffd6b80 0x40103f33:0x3ffd6be0 0x401034b6:0x3ffd6c20 0x4010462b:0x3ffd6c40 0x40101c73:0x3ffd6d80 0x400ea1a7:0x3ffd6db0 0x400d53be:0x3ffd6dd0 0x400f6831:0x3ffd6df0 0x400dd815:0x3ffd6e20 0x4012453d:0x3ffd6e40 

  #0  0x40084145:0x3ffd68c0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
  #1  0x40095f9d:0x3ffd68e0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
  #2  0x4009bb55:0x3ffd6900 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x401bf0bb:0x3ffd6980 in __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47
  #4  0x401bf102:0x3ffd69a0 in std::terminate() at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57
  #5  0x401bf2dd:0x3ffd69c0 in __cxa_allocate_exception at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_alloc.cc:300
  #6  0x401bf398:0x3ffd69e0 in operator new(unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54
  #7  0x400fc219:0x3ffd6a00 in __gnu_cxx::new_allocator<std::_Sp_counted_ptr_inplace<json11::JsonObject, std::allocator<json11::JsonObject>, (__gnu_cxx::_Lock_policy)2> >::allocate(unsigned int, void const*) at c:\users\thomas\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\ext/new_allocator.h:111
      (inlined by) std::allocator_traits<std::allocator<std::_Sp_counted_ptr_inplace<json11::JsonObject, std::allocator<json11::JsonObject>, (__gnu_cxx::_Lock_policy)2> > >::allocate(std::allocator<std::_Sp_counted_ptr_inplace<json11::JsonObject, std::allocator<json11::JsonObject>, (__gnu_cxx::_Lock_policy)2> >&, unsigned int) at c:\users\thomas\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/alloc_traits.h:436
      (inlined by) std::__allocated_ptr<std::allocator<std::_Sp_counted_ptr_inplace<json11::JsonObject, std::allocator<json11::JsonObject>, (__gnu_cxx::_Lock_policy)2> > > std::__allocate_guarded<std::allocator<std::_Sp_counted_ptr_inplace<json11::JsonObject, std::allocator<json11::JsonObject>, (__gnu_cxx::_Lock_policy)2> > >(std::allocator<std::_Sp_counted_ptr_inplace<json11::JsonObject, std::allocator<json11::JsonObject>, (__gnu_cxx::_Lock_policy)2> >&) at c:\users\thomas\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/allocated_ptr.h:97
      (inlined by) std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<json11::JsonObject, std::allocator<json11::JsonObject>, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, json11::Json, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, json11::Json> > > >(json11::JsonObject*&, std::_Sp_alloc_shared_tag<std::allocator<json11::JsonObject> >, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, json11::Json, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, json11::Json> > >&&) at c:\users\thomas\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/shared_ptr_base.h:675
      (inlined by) std::__shared_ptr<json11::JsonObject, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<json11::JsonObject>, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, json11::Json, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, json11::Json> > > >(std::_Sp_alloc_shared_tag<std::allocator<json11::JsonObject> >, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, json11::Json, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, json11::Json> > >&&) at c:\users\thomas\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/shared_ptr_base.h:1342
      (inlined by) std::shared_ptr<json11::JsonObject>::shared_ptr<std::allocator<json11::JsonObject>, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, json11::Json, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_s [... cut ...]
      (inlined by) WebServerThread::runOnce() at src/mesh/http/WebServer.cpp:177
  #17 0x400d53be:0x3ffd6dd0 in concurrency::OSThread::run() at src/concurrency/OSThread.cpp:78
  #18 0x400f6831:0x3ffd6df0 in ThreadController::runOrDelay() at .pio/libdeps/tlora-v2-1-1.6/Thread/ThreadController.cpp:59
  #19 0x400dd815:0x3ffd6e20 in loop() at src/main.cpp:513
  #20 0x4012453d:0x3ffd6e40 in loopTask(void*) at C:/Users/thomas/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 0000000000000000

Rebooting...
```

caveman99, 2022-10-22:
I merged the fix for now - still need to adress the dump.

sachaw, 2022-10-22:
> > I've always struggled to even get the CSV file to be created
> 
> It has been elusive

Just tested and got it working fine, try the latest build

caveman99, 2022-11-23:
JSON lib swapped out for 2.0.5 - no more leaky.

closed 2022-11-23 by caveman99