meshtastic/firmware #1837: [modules][canned messages] add ability to specify destination for canned messages

lesykm, 2022-10-22:
i.e. send not to broadcast channel but also direct message to specific node.
Perhaps besides `message` field (https://github.com/meshtastic/Meshtastic-protobufs/blob/d0559bfa3c31023ed2f2aa3807b6a0a1da9a6feb/cannedmessages.proto#L16) also `dest` field can be added there where name or id of the node can be specified to send message to.

caveman99, 2022-10-22:
The CardKB implementation already does this. Press tab and select a destination node for DM. For the rotary encoders it's mainly a GUI problem.

lesykm, 2022-10-22:
This happens only for freetext form: https://github.com/meshtastic/Meshtastic-device/blob/master/src/modules/CannedMessageModule.cpp#L241

What I suggest is for canned (pre-saved) messages which currently supports only broadcast: https://github.com/meshtastic/Meshtastic-device/blob/master/src/modules/CannedMessageModule.cpp#L249

I'll try to prepare PRs this week to implement this

caveman99, 2022-10-22:
yes, the target selection is only working in freetext mode, but the idea was to bring that to the canned mode too. I was contemplating different ways to implement it with only 3 buttons. PR's gladly welcome :-)

Matrix Keyboards are already supported in the code for the special case of the RAK 3x3 and 4x4 modules. They're conencted by I2C and immediately send the message in the slot of the ordinace key pressed. so pressing key 3 sends the 3rd message. Not sure how this is behaving when no screen is detected.

garthvh, 2023-08-09:
Addressed in https://github.com/meshtastic/firmware/issues/2590

closed 2023-08-09 by garthvh

ulrichard, 2024-02-12:
On the t-watch with firmware 2.2.19 canned messages are always broadcasted, regardless of the selected recipient.
If I send a free text message from the web ui connected through serial, it is only delivered to the selected recipient.

ulrichard, 2024-02-18:
I just had a quick look at the source code, and I am wondering why there is a NODENUM_BROADCAST hardcoded at https://github.com/meshtastic/firmware/blob/143ee9cdf61773f0d719a9162affd96427f8b3ad/src/modules/CannedMessageModule.cpp#L305 when destination selection is handled at https://github.com/meshtastic/firmware/blob/143ee9cdf61773f0d719a9162affd96427f8b3ad/src/modules/CannedMessageModule.cpp#L345