meshtastic/firmware #1848: [Bug]: Pressing too fast on the middle button restarts the device

MihaiRF, 2022-10-22:
### Category

Other

### Hardware

T-Beam

### Firmware Version

1.3.46.d4ea956

### Description

Pressing too many times (too fast) on the middle button (for changing between screens) is causing a device reset.

thebentern, 2022-10-22:
I think I might have observed this behavior in the past. Would you mind getting some serial logs via meshtastic --noproto when this occurs?

MihaiRF, 2022-10-23:
So,
If I press once:
[image]

If I press twice (quickly):
[image]

If I press three times (quickly):
[image]
and it reboots...

thebentern, 2022-10-23:
This is actually a _feature_ that I had forgotten about. Unfortunately I can't find any documentation about it other than a thread on the discourse group when it was implemented originally. A fast triple click is set up to clear the device bluetooth bonds and restart the device. This was likely more useful when our bluetooth stack was a bit less stable in previous versions.
@garthvh where should we put this in the docs? Or I'm also totally open to revisting this functionality as well... I'm not particularly fond of our current user button scheme. I'm not convinced anyone actually uses the long-press oled brightness toggle for instance.

closed 2022-10-23 by thebentern

garthvh, 2022-10-23:
There is a YouTube video

MihaiRF, 2022-10-23:
I didn't know about the Oled brightness functionality, cool thing :)
As for the triple click feature, is not really a problem once you know about it. I would leave it as it is, at least for the time being. 
Thanks.

thebentern, 2022-10-23:
If you're in the Discord group, we're actually talking about re-evaluating some of these user-button functions and switching them to more useful options in light of the evolution of the firmware / apps that's occurred since they were originally added.
https://discord.com/invite/UQJ5QuM7vq

MihaiRF, 2022-10-23:
> If you're in the Discord group in the #meshtastic-1_3 room, we're actually talking about re-evaluating some of these user-button functions and switching them to more useful options in light of the evolution of the firmware / apps that's occurred since they were originally added. https://discord.com/invite/UQJ5QuM7vq

Thanks for the invitation. I'll try to follow as my time allows.