meshtastic/firmware #1849: [Feature Request] Implement morse function to external notification module

Bergstern, 2022-10-23:
We have the possibility to light up a LED or to activate a buzzer with the ext. notification module. So my idea is to implement a second ASCII Charakter to send Morse code to a node, this way it would be possible to send morse code (audio/visual) via Meshtastic. 
There must be two different "external_notification.output_ms" for dot and dahh (./-)

Perhaps there is chance to set a channel to "Morse/CW Mode". If "Morse/CW Mode" is true the received message will translate into morse code and turn on the gpio output to high/low till the message ends.

The other possibility would be that if "Morse/CW  Mode" is active the receiving node controlls incoming messages if morse code is included.  (I think at a nomal conversation nobody writes like this ".... . -.. .-.. --- "  ;-) ) 

I know that a feature like this is not for everyone a must have. But i think it would be a cool feature where people could learn CW or simply play around with it.

Thank u for ur attention
Andre

sachaw, 2022-10-24:
Not sure what the point would be other than just because, as using a cw would defeat the point of having LoRa in the first place. And the encapsulation of the mouse would be digital so not even real mouse code. If you have another approach, I'm all ears

garthvh, 2022-11-26:
@Bergstern closing this issue as this is probably not the preferred approach, can you submit a pull request for us to review with your new morse channel?

closed 2022-11-26 by garthvh

mashu, 2025-04-02:
Adding Morse code to Meshtastic would give us a practical Vband/Morserino-like experience right on our existing mesh network - convenient for skills practice without internet dependency or extra hardware. Many HAMs would appreciate this. The ability to connect a paddle directly to a node or via a simple dongle combines genuine utility with the tactile enjoyment many hams appreciate. It's this blend of practicality and fun that makes the idea compelling - maintaining Morse proficiency while adding a yet functional communication option to our modern mesh setup. I'd personally find this valuable both for keeping my skills sharp and for the simple satisfaction of tapping out messages across the network. It's definitely worth adding especially that people use things like that (i.e Vband).