meshtastic/firmware #1857: [Bug]: MQTT URL ignores appended port number

BlipRanger, 2022-10-25:
### Category

WiFi, Other

### Hardware

Not Applicable

### Firmware Version

1.3.47.05147c0-d

### Description

When attempting to set a url for the MQTT server, the port appended to the end is ignored. This might be intentional, but some servers use non-standard ports for MQTT and it would be nice if we could specify in the normal fashion  ("my.server.com:1885") rather than assuming that it will always be 1883.

caveman99, 2022-10-25:
please check again, cause this firmware code will do exactly what you want: https://github.com/meshtastic/Meshtastic-device/blob/761804b17a5021c7cb67e07a3e89f0a0c15f72e0/src/mqtt/MQTT.cpp#L131-L139

closed 2022-10-25 by BlipRanger

sachaw, 2022-10-26:
I'll look I to this.