meshtastic/firmware #1869: [Bug]: MQTT JSON TEXT_APP not contain original message source information

D4rk4, 2022-10-28:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

1.3.47

### Description

TEXT_APP JSON output to MQTT did not contain any information about the message source.
Only the sender node (connected to MQTT) is described in the message. I guess the sender node ID must be placed in the payload.id field like it's already implemented for nodeinfo.

### Relevant log output

```Shell
2022/10/29 00:53:54 Received message msh/2/json/LongFast/!25053ad4
2022/10/29 00:53:54 Payload {"channel": 0, "from": -228157708, "id": 397733322, "payload": {"hardware": 2, "id": "!f26696f4", "longname": "nf404", "shortname": "404"}, "sender": "!25053ad4", "timestamp": 1666994034, "to": -1, "type": "nodeinfo"}
2022/10/29 00:59:11 Received message msh/2/json/LongFast/!25054758
2022/10/29 00:59:11 Payload {"channel": 0, "from": -1277786198, "id": -2034113244, "payload": {"text": "T"}, "sender": "!25054758", "timestamp": 1666994350, "to": -1, "type": "text"}
```

arduionoGP, 2022-10-29:
(you could incorporate any identifying data you wish to send directly into that text message payload as JSON from the sending radio node)For using mqtt to merge two hub and spoke type topologies over a network into one it might be beneficial to identify the specific radio the data came from instead of the one acting as a hub to the broker.
Yes, your payloads look like the ones I am receiving.
TEXT_MESSAGE_APP msh/2/json
{"channel": 0, "from": 2130624652, "id": -852747030, "payload": {"text": "Testing"}, "sender": "!7efeb2c8", "timestamp": 0, "to": -1, "type": "text"}
NODEINFO_APP msh/2/json
{"channel": 0, "from": 2130621128, "id": 845534553, "payload": {"hardware": 10, "id": "!7efeb2c8", "longname": "Meshtastic b2c8", "shortname": "b2c8"}, "sender": "!7efeb2c8", "timestamp": 0, "to": -1, "type": "nodeinfo"}

garthvh, 2022-10-31:
With flood routing there is no trace route like functionality, messages have just a to and a from.

All of the available packet fields are serialized to JSON or sent to MQTT unmodified, there is no adding or removing of fields on the device side.

closed 2022-10-31 by garthvh

D4rk4, 2022-10-31:
@garthvh so, for now, is now FROM field in JSON!