meshtastic/firmware #1877: [Bug]: MQTT downlink message to device stopped working with 1.3.48

arduionoGP, 2022-10-31:
### Category

Other

### Hardware

T-Beam, Heltec v2.1

### Firmware Version

1.3.48.82bcd39

### Description

Was testing the new alpha release and first noted that client.meshtastic.org Channel Editor would silently fail to save Uplink and Downlink Enabled to device. It seemed to work with the CLI so I reported it as an issue on the Meshtastic-Web issues.

I then tested sending JSON envelope from MQTT to the devices. It had worked in 1.3.46. The same message fails to be noted and rebroadcast by the device. It also does not get sent back out to MQTT broker from the device on the cleartext channel msh/2/c/#.

The sent message was of the form { "sender": "Broker, "payload": "Here is an injected text message going to be sent by radio."}. It was sent on topic: "msh/2/json/LongFast/!gatewayID".  This was based on the documentation of a valid envelope.

arduionoGP, 2022-10-31:
Nevermind. I figured the issue out. MQTT.cpp if statement on line 44.

A valid envelope now needs to have an additional item "type": "sendtext" added to it.

When I changed the message to: {"sender":"Broker, "type":"sendtext", "payload":"Here is an injected message."}
It works just fine.

arduionoGP, 2022-10-31:
Closing, I will need to update the docs.

closed 2022-10-31 by arduionoGP

caveman99, 2022-10-31:
sorry about that, that was a side change i forgot to document. I made it to not 'close loop' on json, but only on the binary protobuf version of the messages. JSON is meant to extract and inject stuff, not interlink nodes...