meshtastic/firmware #1881: [Bug]: [modules][external notifications] external notifications listens for bells in gpio channel while canned message sends them in another one

lesykm, 2022-10-31:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.0.0.d978c01c-d

### Description

As titled.

caveman99, 2022-11-01:
valid point, but i'd like to discuss this further. Rather than adding just another setting to the moduleconfig, can we alter the behaviour of the module being bound to the GPIO channel?

I found something similar when i added text message support to the serial module. the module in its former state would only communicate on a channel called serial, like the admin module would only talk on a channel named admin and this one is GPIO only. This was introduced before my time, but i think was just copied over from the admin channel implementation, to 'encapsulate' traffic and shield it from the usual 'visible' messaging. I changed it, so that the sender would be bound to the channel name only if the payload was not a textmessage. Ane asy solution here would be to make the notification RECEIVER not be bound to a channel name at all, becuase the BELL character is so special that i won't be included in another Text Message payload by mistake. 

That would spare us another definition users are asking questions about and will enable the desired and logical functionality. Thoughts please?

lesykm, 2022-11-02:
Easy, it is commenting only one line then.
However that will lead to 2 issues:
1. It is change in behavior (may be it's OK?)
2. `alert_message` = true property will lead to module trigger on any message on any channel and not only on bell message (which is separate property (`alert_bell` = true)) 

For my usecase this is absolutely not an issue and if you are OK with commenting of bound channel line I'll gladly update attached PR :)

caveman99, 2022-11-02:
can we somehow decide which method is active and bind/unbind on this condition? In the serial module i am doing something similar while initializing. When changing config, the node is rebooted anyway.

so if alert_message is true, bind to the channel name, for all other conditions, don't ... This will keep behaviour for 'all messages' intact and register the bell in all textmessages. (unless alert_message is specified)

closed 2022-11-03 by caveman99