meshtastic/firmware #1883: [Bug]: setting admin channel breaks connectivity towards other devices

IZ1IVA, 2022-11-01:
### Category

Other

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.0.0

### Description

After a fresh install, setting up the same admin channel results in the devices being unable to join the mesh.
`meshtastic --info` confirms the channel URL is the same for all devices, their displays show "Custom-X" as their channel.

It was working in 1.3.47.

garthvh, 2022-11-01:
I can't replicate this, we have a two channel mesh working with 5 devices on 2.0, are all your devices running 2.0?  We need the --info or a serial log showing the issue.

garthvh, 2022-11-01:
Please test with a preset for now

closed 2022-11-01 by garthvh

IZ1IVA, 2022-11-01:
Did some more testing.
Yes, all devices run on 2.0.0, but the admin channel was created on a previous version (perhaps1.3.46).
Though remote admin is still hit-and-miss, I managed to get it working by creating a new admin channel (i.e. starting afresh).

Thank you for your support, cheers!