meshtastic/firmware #1889: [Bug]: T-Beam with M8N & SX1262 with the 1.3" OLED shuts down but the text of the last screen displayed stays on the screen and garbage on the screen.

dmg244, 2022-11-02:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.0.0.18ab874

### Description

T-Beam with M8N & SX1262 with the 1.3" OLED shuts down but the text of the last screen displayed stays on the screen and garbage on the screen. firmware-v2.0.0.18ab874

garthvh, 2022-11-02:
This has been fixed recently, is not perfect https://github.com/meshtastic/firmware/pull/1876

thebentern, 2022-11-02:
Worth mentioning... if you shutdown from one of the apps or cli, it will reliably because we control the shutdown sequence. With the long-press PWR button shutdown, the power is cut by the power management chip which is hard to "catch" in the firmware from a timing perspective.

dmg244, 2022-11-02:
> This has been fixed recently, is not perfect #1876

It's is fixed in firmware-v2.0.0.18ab874 or not?

dmg244, 2022-11-03:
I'm understand why screen not switch off. But what with garbage on the screen? Last working firmware without this problem is 1.3.42.9bd9252. In next version 1.3.44.4fa8d02 and later 1.3" is not working.

garthvh, 2022-11-03:
It is fixed as much as we control, the tbeam power button can cut power before we can do anything. The 1.3 inch screw never turned off with the tbeam power button

dmg244, 2022-11-03:
I want say that OLED 1.3 stop working after firmware-1.3.41.80ddb81.
[image]

dmg244, 2022-11-03:
This is v1.3.41.80ddb81
[image]

dmg244, 2022-11-03:
Small OLED screen working perfect.

dmg244, 2022-11-03:
Standalone screens 1.3 and small on fw v2:
[image]
[image]

garthvh, 2022-11-03:
[image]

garthvh, 2022-11-03:
It was never supported on 1,2

dmg244, 2022-11-03:
I don't understand, what  "never supported on 1,2" ?

garthvh, 2022-11-03:
The 1.3" TTGO 3 button ssh1106 screen was not supported in 1.2 without turning the resistor on the back of the screen

dmg244, 2022-11-03:
I think you display and my display have different controller. I buyOLED board two week ago. Mine OLED display have SH1106. This controller more cheaper and now manufacture use them. You display have SSD1306. I repeat my 1.3 OLED screen work fine before fw v.1.3.42.9bd9252:

 firmware-1.3.41.80ddb81 - working fine
 firmware-v.1.3.42.9bd9252 - NOT working
....
....
firmware-v2.0.0.18ab874 - still NOT working

Something changed from version 1.3.41.80ddb81

mc-hamster, 2022-11-03:
> I don't understand, what  "never supported on 1,2" ?

That means that we did not build in explicit support for that device. If it worked, it was out of pure luck. We're a group of volunteers who pitch in time in our off hours. We don't have a test bench of all the hardware that the world produces. Pretty much, open source contributors do what they are personally passionate with.

If this is something you are passionate about, we'll gladly help you (or people you know) be familiar with our code so you can add support for this. 

Cheers!

caveman99, 2022-11-03:
The LilyGo 1.3 screen is working for me as well in 2.0, the controller (1106 or 1306) is autodetected as good as possible. 

Is this garbage on screen right away on bootup or will it fade over time?

Can you post a serial bootleg of the first few seconds after you switch on the device? 

Will the screen work if you press 'Reset' when it shows garbage?

dmg244, 2022-11-03:
Garbage on screen right away from bootup.
Reset don't help.

??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 axp192/axp2101 PMU found
??:??:?? 0 I2C device found at address 0x3c
??:??:?? 0 **0x6 subtype probed in 2 tries**
??:??:?? 0 ssd1306 display found
??:??:?? 0 2 I2C devices found
??:??:?? 0 Meshtastic hwvendor=4, swver=**2.0.0.18ab874-d**

??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 axp192/axp2101 PMU found
??:??:?? 0 I2C device found at address 0x3c
??:??:?? 0 **0x3 subtype probed in 2 tries**
??:??:?? 0 ssd1306 display found
??:??:?? 0 2 I2C devices found
??:??:?? 0 Meshtastic hwvendor=4, swver=**1.3.41.80ddb81-d**

dmg244, 2022-11-03:
Modification OLED 1.3 screen to power from AXP192:
[image]
[image]

caveman99, 2022-11-03:
Looks like you have one of the few Displays that is detected wrong. Despite probing 6 times it decides it's a 1306. There will be a possibility to override that detection soon.

dmg244, 2022-11-03:
I'ts allow to switch off 1.3" screen like 0.96" from power button.

dmg244, 2022-11-03:
Now I try to compile fw with option **#define USE_SH1106**
Never do it before... Try install PlatformIO and etc...

caveman99, 2022-11-03:
No need to, I'll do a new version within 24 hours :-)

dmg244, 2022-11-03:
It's just interesting for me. I have little bit free time now and want try it. ;-)

caveman99, 2022-11-03:
Don't get me wrong, don't wanna discourage you, try out PIO you will find it rather easy to produce your custom firmware with it. Maybe we'll see your first PR soon :-)

dmg244, 2022-11-03:
What is PIO?

dmg244, 2022-11-03:
Finally with 1.3" OLED board modification and fw-2.0.0 with **#define USE_SH1106** all work fine. Will be wait new firmware for testing new autodetect screen procedure.
[https://youtu.be/wO47Hgglw00](video)

caveman99, 2022-11-03:
the new firmware is here https://github.com/meshtastic/firmware/suites/9119913152/artifacts/423342217 - however it is not a fix for the autodetect, you can switch autodetect off with it. You can't use it immediately, it needs a new CLI version to make the setting available. Once the new CLI is released, you caan do `meshtastic --set screen.oled OLED_SH1106`

closed 2022-11-17 by caveman99

hickey, 2022-11-18:
I have a nearly identical problem to the one described here. The hardware is the same t-beam with M8N and a SX1262. I have the 1.3" OLED display with 3 buttons. I originally loaded firmware 2.0.0-18ab874 and my display showed a completely random pattern displayed like this [image](https://user-images.githubusercontent.com/33575781/199706094-2e7651d2-10e0-4e2c-a894-373a785db0a7.jpg). The text was offset and wrapped on the display. 

I just tried the firmware listed above and the display improved, but is still useless. 

[image]

[image]

Here is the boot sequence:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   459][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
��@??:??:?? 0

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files (475136/1048576 Bytes):
??:??:?? 0  /prefs/channels.proto (53 Bytes)
??:??:?? 0  /prefs/db.proto (143 Bytes)
??:??:?? 0  /static/.gitkeep (0 Bytes)
??:??:?? 0  /static/Logo_Black.svg.gz (592 Bytes)
??:??:?? 0  /static/icon.svg.gz (842 Bytes)
??:??:?? 0  /static/index.2759ba5d.js.gz (347 Bytes)
??:??:?? 0  /static/index.85831067.css.gz (15140 Bytes)
??:??:?? 0  /static/index.d4fbecc1.js.gz (410906 Bytes)
??:??:?? 0  /static/index.html.gz (507 Bytes)
??:??:?? 0  /static/robots.txt.gz (42 Bytes)
??:??:?? 0  /static/site.webmanifest.gz (196 Bytes)
[   730][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 axp192/axp2101 PMU found
??:??:?? 0 I2C device found at address 0x3c
??:??:?? 0 0x0 subtype probed in 1 tries
??:??:?? 0 sh1106 display found
??:??:?? 0 2 I2C devices found
??:??:?? 0 Meshtastic hwvendor=4, swver=2.0.0.6c07fbf-d
??:??:?? 0 Setting random seed 3276851166
??:??:?? 0 Total heap: 197044
??:??:?? 0 Free heap: 163384
??:??:?? 0 Total PSRAM: 4192123
??:??:?? 0 Free PSRAM: 4191863
??:??:?? 0 NVS: UsedEntries 70, FreeEntries 560, AllEntries 630, NameSpaces 3
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 6
??:??:?? 0 OTA firmware version 0.1.1.98ab02e
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 20
```

When I purchased the displays on Aliexpress it was reportedly a SH1106 controller.