meshtastic/firmware #1891: [Bug]:

he-leon, 2022-11-02:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

from 1.3.429bd9252  to 2.0.0.18ab874 

### Description

Hi,
I'm having difficulties to get newer versions of meshtastics running on my ttgo lora32 v2 hardware.  

The last working on my device is 1.3.41.  The following releases (1.3.429bd9252 - 2.0.0.18ab874) result in a reset loop of the ESP32 as can be seen in the serial log:

```
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
�� @??:??:?? 0 

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=intWatchdog
```

Is this a known problem? Can someone confirm this issue?
Also, I couldn't select ttgo lora32 v2 in the hardware dropdown for this issue. Does this mean this board is not supported anymore?

Cheers

caveman99, 2022-11-03:
your board is supported, try doing an install instead of an upgrade. (Install will erase flash, something is tangled up there)

j4cko, 2022-11-03:
I have the same issue and tried `device-install.sh -f firmware-tlora-v2-2.0.0.18ab874.bin` (there was no meshastic installed previously). Downgrading to 1.3.41 works.

he-leon, 2022-11-03:
@caveman99 thanks for the quick response. 
I used the ESP32 web installer and choose "wipe and reinstall". This results in the reset loop as explained above.

caveman99, 2022-11-03:
This may be the same issue we had with T-Lora 1.0 since the 2.0 board also uses the old revision of the CPU. The 1.3 and 2.0 boards are not in wide circulation, so this may have gone under the radar since 1.3.41. I'll post a possible fix later tonight

caveman99, 2022-11-03:
can you try the firmware from https://github.com/meshtastic/firmware/suites/9119918329/artifacts/423343200 ? This should run again on the tlora 2

j4cko, 2022-11-04:
Thanks for your quick efforts. Unfortunately, it does not improve things for me, I am still getting the same output on the serial as @he-leon above.

he-leon, 2022-11-04:
I can confirm this: Same reset loop with firmware-tlora-v2-2.0.0.8ce7b5c.bin.
My chip revision is ESP32-PICO-D4 (revision v1.0)

he-leon, 2022-11-04:
I experienced the same behavior (reset loop) on an other esp32 sketch using the SSD1306 OLED.
The problem was caused by the following lines in `setup()`:

```
void setup() {
  //reset OLED display via software
  pinMode(OLED_RST, OUTPUT);
  digitalWrite(OLED_RST, LOW);
  delay(20);
  digitalWrite(OLED_RST, HIGH);

....
```

Removing the reset procedure fixed the issue.

Interestingly the problem only occurs on recent versions of the espressif32 platform. 
For example, the above code works fine when using:
```
platform = espressif32@3.4.0
```

Seemingly meshtastic uses the same reset procedure on startup. I submitted a PR which disables the reset by removing the Reset pin definition.
See #1898

caveman99, 2022-11-04:
thanks for your contribution. I remember now this is how we finally fixed it on the 1.0 boards as well. the GPS pins were a red herring. Pulled the PR into Master...

closed 2022-11-04 by caveman99