meshtastic/firmware #1913: [Bug]: WDT timeout when leaving light sleep by button press

weber-software, 2022-11-07:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.0.b7ef6323-d

### Description

when the device is in light sleep i am unable to wake it by the button.

It seems like it is going to "DARK" instead of "ON" and is going to reboot by WDT timeout once i press the button a little longer.

Or it is rebooting immediately while the button is pressed.

I assume that it is depending on the timing - but anyway i would expect that it goes to "ON" and enables the display.

### Relevant log output

```Shell
??:??:?? 234 [PowerFSM] lsEnter begin, ls_secs=300

??:??:?? 234 [PowerFSM] Disable bluetooth

??:??:?? 234 [PowerFSM] GPS prepare sleep!

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:

PC      : 0x40081a58  PS      : 0x00060935  A0      : 0x40086af4  A1      : 0x3ffbf92c  

A2      : 0x00000040  A3      : 0x3ffbe3d0  A4      : 0x00000001  A5      : 0x4009727e  

A6      : 0x00060623  A7      : 0x00060623  A8      : 0x80081aed  A9      : 0xffffff00  

A10     : 0x00000040  A11     : 0x00000020  A12     : 0x00000001  A13     : 0x3ffd68d0  

A14     : 0x0e41e0b5  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  

EXCVADDR: 0x00000000  LBEG    : 0x400865d9  LEND    : 0x400865e1  LCOUNT  : 0x00000027  

Core  1 was running in ISR context:

EPC1    : 0x401367f7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x40081a55:0x3ffbf92c |<-CORRUPTED

Core  0 register dump:

PC      : 0x4020487a  PS      : 0x00060a35  A0      : 0x801351a8  A1      : 0x3ffbd6c0  

A2      : 0x00000000  A3      : 0x80000001  A4      : 0x80090471  A5      : 0x3ffbf2ec  

A6      : 0x00000000  A7      : 0x3ff4e0c4  A8      : 0x80134ce6  A9      : 0x3ffbd690  

A10     : 0x00000000  A11     : 0x80000001  A12     : 0x8009ab24  A13     : 0x3ffbd5c0  

A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  

EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x40204877:0x3ffbd6c00x401351a5:0x3ffbd6e0 0x400987b0:0x3ffbd700 

ELF file SHA256: 0000000000000000

Rebooting...
```

weber-software, 2022-11-07:
I guess it it related:

Somehow the `bool pressed = !digitalRead(BUTTON_PIN);` in `lsEnter` is also not working correctly.

There is no difference when waking up by receiving a packet or by the button press. In both cases it is recognized as `EVENT_WAKE_TIMER`.
In case of the packet receiving everything is working as i would expect.

Wakeup by button press:

```
??:??:?? 431 [PowerFSM] lsEnter begin, ls_secs=300
??:??:?? 431 [PowerFSM] Disable bluetooth
??:??:?? 431 [PowerFSM] GPS prepare sleep!
??:??:?? 457 [PowerFSM] Exit light sleep gpio: btn=0
??:??:?? 457 [PowerFSM] wakeCause2 7
??:??:?? 457 [PowerFSM] EVENT_WAKE_TIMER
Transition powerFSM transition=Wake timer, from=LS to=DARK
??:??:?? 457 [PowerFSM] Exit state: LS
??:??:?? 457 [PowerFSM] Allowing GPS lock
??:??:?? 457 [Power] Battery: usbPower=1, isCharging=1, batMv=3857, batPct=63
```

Wakeup by incoming packet.
```
Transition powerFSM transition=Bluetooth timeout, from=DARK to=LS
??:??:?? 340 [PowerFSM] lsEnter begin, ls_secs=300
??:??:?? 340 [PowerFSM] Disable bluetooth
??:??:?? 340 [PowerFSM] GPS prepare sleep!
??:??:?? 359 [PowerFSM] Exit light sleep gpio: btn=0
??:??:?? 359 [PowerFSM] wakeCause2 7
??:??:?? 359 [PowerFSM] EVENT_WAKE_TIMER
Transition powerFSM transition=Wake timer, from=LS to=DARK
??:??:?? 359 [PowerFSM] Exit state: LS
??:??:?? 359 [PowerFSM] Allowing GPS lock
??:??:?? 359 [Power] Battery: usbPower=1, isCharging=1, batMv=3855, batPct=63
??:??:?? 359 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=66, time 1148 ms
??:??:?? 359 [RadioIf] Lora RX (id=0x282b193e Fr0xf0 To0xe4, WantAck0, HopLim3 Ch0xc0 encrypted rxSNR=7 rxRSSI=1.84067)
??:??:?? 359 [RadioIf] AirTime - Packet received : 1148ms
??:??:?? 359 [Router] Add packet record (id=0x282b193e Fr0xf0 To0xe4, WantAck0, HopLim3 Ch0xc0 encrypted rxSNR=7 rxRSSI=1.84067)
```

caveman99, 2022-11-07:
Can you please post a log of the first 60 seconds after bootup?

weber-software, 2022-11-07:
sure:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   460][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
��@??:??:?? 0 

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files (16384/1048576 Bytes):
??:??:?? 0  /prefs/channels.proto (91 Bytes)
??:??:?? 0  /prefs/config.proto (84 Bytes)
??:??:?? 0  /prefs/db.proto (144 Bytes)
[   536][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 axp192/axp2101 PMU found
??:??:?? 0 I2C device found at address 0x3c
??:??:?? 0 0x4 subtype probed in 2 tries 
??:??:?? 0 ssd1306 display found
??:??:?? 0 2 I2C devices found
??:??:?? 0 Meshtastic hwvendor=4, swver=2.0.0.0832cc50-d
??:??:?? 0 Setting random seed 1216351547
??:??:?? 0 Total heap: 197028
??:??:?? 0 Free heap: 163232
??:??:?? 0 Total PSRAM: 4192123
??:??:?? 0 Free PSRAM: 4191863
??:??:?? 0 NVS: UsedEntries 91, FreeEntries 539, AllEntries 630, NameSpaces 3
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 53
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4009722f
file: "src/platform/esp32/BleOta.cpp" line 15
func: static const esp_partition_t* BleOta::findEspOtaAppPartition()
expression: esp_ota_get_partition_description(part, &app_desc)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4009722f
file: "src/platform/esp32/BleOta.cpp" line 28
func: static String BleOta::getOtaAppVersion()
expression: esp_ota_get_partition_description(part, &app_desc)
??:??:?? 0 No OTA firmware available
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 20
??:??:?? 0 Loading /prefs/config.proto
??:??:?? 0 Loaded saved config version 20
[   695][E][vfs_api.cpp:104] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
??:??:?? 0 No /prefs/module.proto preferences found
??:??:?? 0 Installing default ModuleConfig
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Loaded saved channelFile version 20
[   731][E][vfs_api.cpp:104] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
??:??:?? 0 No /oem/oem.proto preferences found
??:??:?? 0 Number of Device Reboots: 53
??:??:?? 0 Wanted region 3, using EU_868
??:??:?? 0 region=3, NODENUM=0xa3fb45e4, dbsize=1
??:??:?? 0 Saving /prefs/db.proto
[   846][I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
[   846][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
??:??:?? 0 Warning: Failed to find AXP2101 power management
[   858][I][XPowersAXP2101.tpp:230] ~XPowersAXP2101(): ~XPowersAXP2101
[   859][I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
[... 141 lines cut ...]
??:??:?? 36 [Router] Received routing from=0xa3fb58f0, id=0x4d256b7f, portnum=4, payloadlen=44
??:??:?? 36 [Router] Routing sniffing (id=0x4d256b7f Fr0xf0 To0xe4, WantAck0, HopLim3 Ch0x0 Portnum=4 requestId=50e65a8b rxSNR=9.75 rxRSSI=1.84067)
??:??:?? 36 [Router] Received an ack for 0x50e65a8b, stopping retransmissions
??:??:?? 36 [Router] FIXME-update-db Sniffing packet
??:??:?? 36 [Router] Delivering rx packet (id=0x4d256b7f Fr0xf0 To0xe4, WantAck0, HopLim3 Ch0x0 Portnum=4 requestId=50e65a8b rxSNR=9.75 rxRSSI=1.84067)
??:??:?? 36 [Router] Forwarding to phone (id=0x4d256b7f Fr0xf0 To0xe4, WantAck0, HopLim3 Ch0x0 Portnum=4 requestId=50e65a8b rxSNR=9.75 rxRSSI=1.84067)
??:??:?? 36 [Router] Update DB node 0xa3fb58f0, rx_time=0
??:??:?? 36 [Router] Module 'routing' considered
??:??:?? 36 [Screen] Turning on screen
??:??:?? 37 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=66, time 1148 ms
??:??:?? 37 [RadioIf] Lora RX (id=0x50e65a8b Fr0xe4 To0xff, WantAck0, HopLim2 Ch0xc0 encrypted rxSNR=7.25 rxRSSI=1.84067)
??:??:?? 37 [RadioIf] AirTime - Packet received : 1148ms
??:??:?? 37 [Router] Rx someone rebroadcasting for us (id=0x50e65a8b Fr0xe4 To0xff, WantAck0, HopLim2 Ch0xc0 encrypted rxSNR=7.25 rxRSSI=1.84067)
??:??:?? 37 [Router] didn't find pending packet
??:??:?? 37 [Router] Found existing packet record for fr=0xa3fb45e4,to=0xffffffff,id=0x50e65a8b
??:??:?? 37 [Router] Found existing packet record for fr=0xa3fb45e4,to=0xffffffff,id=0x50e65a8b
??:??:?? 37 [Router] Add packet record (id=0x50e65a8b Fr0xe4 To0xff, WantAck0, HopLim2 Ch0xc0 encrypted rxSNR=7.25 rxRSSI=1.84067)
??:??:?? 37 [Router] Ignoring incoming msg, because we've already seen it (id=0x50e65a8b Fr0xe4 To0xff, WantAck0, HopLim2 Ch0xc0 encrypted rxSNR=7.25 rxRSSI=1.84067)
??:??:?? 37 [Router] Incoming message was filtered 0xa3fb45e4
??:??:?? 42 [Power] Battery: usbPower=1, isCharging=1, batMv=3906, batPct=70
Transition powerFSM transition=Screen-on timeout, from=ON to=DARK
??:??:?? 46 [PowerFSM] Turning off screen
??:??:?? 62 [Power] Battery: usbPower=1, isCharging=1, batMv=3906, batPct=70
??:??:?? 82 [Power] Battery: usbPower=1, isCharging=1, batMv=3906, batPct=70
```

caveman99, 2022-11-07:
thanks ill try to debug this later on.

weber-software, 2022-11-07:
one additional finding:
- light sleep
- receive a message
- in `ON` state
- Button is not reacting! Longer press triggers the WDT timeout also.

weber-software, 2022-11-07:
commenting out `wakeOnIrq(BUTTON_PIN, FALLING);` in ButtonThread.h` seems to fix the problem... but is this a valid solution?

garthvh, 2023-01-26:
Is this still happening?

closed 2023-03-26 by garthvh