meshtastic/firmware #1914: [Bug]: File rangetest.csv is not written during the rangetest

svenna1980, 2022-11-07:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

2.0.0.18ab874-d

### Description

it is assumed that the directory static does not exist and therefore File rangetest.csv is not written during the rangetest.

### Relevant log output

```Shell
//\ E S H T /\ S T / C
??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files (16384/1048576 Bytes):
??:??:?? 0 /prefs/channels.proto (53 Bytes)
??:??:?? 0 /prefs/config.proto (125 Bytes)
??:??:?? 0 /prefs/db.proto (419 Bytes)
??:??:?? 0 /prefs/module.proto (24 Bytes)
[ 187][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
??:??:?? 0 I2C device found at address 0x3c
??:??:?? 0 0x4 subtype probed in 2 tries
??:??:?? 0 ssd1306 display found
??:??:?? 0 1 I2C devices found
??:??:?? 0 Meshtastic hwvendor=3, swver=2.0.0.18ab874-d
??:??:?? 0 Setting random seed 19474182
??:??:?? 0 Total heap: 201592
??:??:?? 0 Free heap: 167200
??:??:?? 0 Total PSRAM: 0
??:??:?? 0 Free PSRAM: 0
??:??:?? 0 NVS: UsedEntries 213, FreeEntries 417, AllEntries 630, NameSpaces 7
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 30
??:??:?? 0 OTA firmware version 0.1.1.98ab02e
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 20
??:??:?? 0 Loading /prefs/config.proto
??:??:?? 0 Loaded saved config version 20
??:??:?? 0 Loading /prefs/module.proto
??:??:?? 0 Loaded saved moduleConfig version 20
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Loaded saved channelFile version 20
```

svenna1980, 2022-11-10:
Tested again with 2.0.2. But the issue persists. Under "http://meshtastic.local/static/rangetest.csv" it says "404 Not Found".
http://meshtastic.local/ calls the webinterface. 
[image]
[image]

caveman99, 2022-11-12:
can you take a serial log on the rangetest receiver when it is suppsed to log the entries?

commit 30b6cca3 (2022-11-13): Merge pull request #1925 from meshtastic/bug-1914

fix #1914 and clean up rangetest a bit
commit 4dd140a8 (2022-11-13): fix #1914 and clean up rangetest a bit
closed 2022-11-13 by thebentern