meshtastic/firmware #1928: [Bug]: Negative values in NMEA position sentence

samuelhj, 2022-11-09:
### Category

Serial

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.0.0.18ab874-d

### Description

Using serial module in NMEA mode the sentence reports negative value on longitude. This causes the Garmin device (or any other NMEA0183 capable device) to not understand longitude, usually defaulting to 180°

Sentence example: $GNWPL,6432.48,N,-2207.82,W,BY705*71

But should look like this: $GNWPL,6432.48,N,2207.82,W,BY705*71

samuelhj, 2022-11-09:
This fixed the negative value and the serial module NMEA mode is almost showing correct position ;) . Good work!

closed 2022-11-09 by samuelhj

caveman99, 2022-11-10:
so both the WPL and the GSA (own position) sentences are working as expected?

commit 7da1e5c3 (2022-11-10): Merge pull request #1930 from meshtastic/bug-1928

fix #1928 valid NMEA sentences
commit b004706e (2022-11-10): fix #1928 valid NMEA sentences