meshtastic/firmware #1938: [Bug]: Upgrade Script is Breaking BLE Bonds

garthvh, 2022-11-12:
When running the firmware upgrade script for ESP32 devices BLE bonds on the device are being reset. The expected behavior for an upgrade is for configuration and bonds to remain in place.

caveman99, 2022-11-12:
most likely the new firmware layout is clearing/overwriting the nvs. I'll look into it.

caveman99, 2022-11-12:
Yes, confirmed. We create a new combined binary including the bootloader consisting of those 4 components:

```
    Offset | File
 -  0x1000 | \.pio\build\<board>\bootloader.bin
 -  0x8000 | \.pio\build\<board>\partitions.bin
 -  0xe000 | \.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin
 - 0x10000 | \.pio\build\<board>/firmware.bin
```

The NVS Partition is wedged in between at
```
nvs,      data, nvs,     0x009000, 0x005000,
```
So we need to revise this system and package the components individually. boot_app0.bin is the same for all ESP32 board and variants, bootloader and partitions can be combined into one file per board. so we need to add the boot_app0.bin and anothe file for each ESP32 board to our zip file and change the flashing scripts once more.

caveman99, 2022-11-25:
this will be fixed with the next firmware release

closed 2022-11-25 by caveman99