meshtastic/firmware #1939: [Enhancement]: Include gatewayID as option in protobufs that one might want to send/consume through an MQTT gateway device.

arduionoGP, 2022-11-13:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.3

### Description

nested protobufs (like POSITION_APP, TELEMETRY, TEXT_MESSAGE_APP) passing through the mqtt module get enveloped with a gatewayID that belongs to the MQTT-connnected device and the originating device envelope data is stripped.

In trying to bridge devices on two meshes with mqtt or trying to map lora devices in a hub/spoke fashion by one mqtt device, it would be beneficial to include the gateway of the original device that generates the text or gps data inside the nested packet so the non-mqtt connected originating device is known.

This would require a protobuf update and a touch of code.

garthvh, 2022-11-13:
Why would there be a need to add fields to the protobufs to support MQTT?  Are there fields that are not winding up in the JSON output?

garthvh, 2022-11-13:
For a node I have here the userid is !f71e89f0 which is the hex representation of 4145973744 for example, so you can convert them back and forth to each other, here is the swift that does the conversion

String(format:"%02x", 4145973744) outputs f71e89f0

caveman99, 2022-11-13:
I think we need to rephrase the initial request, it's not really clear to me. 

the 'Envelope' with the Gateway ID is only present on the MQTT Layer and not on the wire. So "the originating device envelope data is stripped" is a misconceoption, cause there is no envelope data. The Meshpacket has a FROM and a TO field and the envelope adds VIA data that is of no concern to the mesh behind it.

What exactly is your intent? do you want to cascade MQTT and have a kind of 'traceroute' of the path the package took? or do you want to use the gateway as a kind of address component for routing messages to a destination mesh? That's what i drew up in February as a concept for internetwork DM's for consideration after 2.0 was ready.

arduionoGP, 2022-11-13:
**Use case one:** Let's say you have one MQTT connected t-beam called !7efec08c. There are 3 other t-beams visible to it via LORA. You want to use MQTT to map or geofence them all in real time using a computer that subscribes to the MQTT messages. All the MQTT packets will come out on the topic msh/2/c/LongFast/!7efec08c.  The payload will contain only the GPS position and !7efec08c as the sender. Once passing through MQTT there is no way of figuring out which device generated those coordinates to plot on the map.

[image]
[image]
[image]

**Use case two:** Let's say you are bridging two LORA meshed groups of devices using one MQTT connected device in each group. Let's say you want to route a message generated from device !7efec08c  on mesh ONE which has MQTT broker !7efeb2c8  over to mesh TWO that has broker !55c731a0. The receiving devices on mesh TWO will have no idea where the message originated from (it will appear to be their MQTT broker) unless there is some source information contained in the message protobuf itself.  

[image]

garthvh, 2022-11-13:
7efeb2c8 = 2130621128
55c731a0 = 1439117728

There is no data that is missing from MQTT that I see here are you expecting TCP like routing information?

caveman99, 2022-11-15:
I think @arduinoGP is really accessing the wrong data fields in this one. I have a private MQTT-Connected Gateway with number 0x25c28464. It's subscribing and publishing on msh/2/c/LongFast/# - now when i get a nodeinfo packet from a radio connected node 0xce0c6a19 i get the following in my (hand crafted script) logfile:

```
Transport Envelope: CHAN LongFast GATE !25c28464
{"from":3456920089,"to":4146001084,"decoded":{"portnum":"NODEINFO_APP","payload":{"id":"!ce0c6a19","longName":"Meshtastic 6a19","shortName":"6a19","macaddr":"0qvODGoZ"},"wantResponse":true},"id":88114534,"rxSnr":13,"priority":-59}
```

Don't look into the payload (it's arbitrary) or the ServiceEnvelope, but the MeshPacket 'from' field and convert that number to hex.

3456920089 = 0xce0c6a19

arduionoGP, 2022-11-16:
I'm sorry, you are absolutely correct. I had missed the conversion where "sender" is one format and "from" is another. This is great! and should making linking and plotting very easy off device.

garthvh, 2022-11-16:
Great!

closed 2022-11-16 by garthvh