meshtastic/firmware #1945: Screen not turning on after recent update

hereiszee, 2022-11-15:
After updating a Tlora-v1 to V2.0.2.8146e84 beta, screen does not turn on. 

Device is working as visible in logs, and is sending packets—screen does not display any data. 

This is especially an issue as i cannot pair the device to my machine and/or phone without the display code.

caveman99, 2022-11-15:
Do you see the screen being detected in the bootup log?

hereiszee, 2022-11-15:
Negative:
`??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Filesystem files (475136/1048576 Bytes):
??:??:?? 0  /prefs/channels.proto (53 Bytes)
??:??:?? 0  /prefs/db.proto (143 Bytes)
??:??:?? 0  /static/.gitkeep (0 Bytes)
??:??:?? 0  /static/Logo_Black.svg.gz (592 Bytes)
??:??:?? 0  /static/icon.svg.gz (842 Bytes)
??:??:?? 0  /static/index.14d6021a.js.gz (410938 Bytes)
??:??:?? 0  /static/index.2759ba5d.js.gz (347 Bytes)
??:??:?? 0  /static/index.85831067.css.gz (15140 Bytes)
??:??:?? 0  /static/index.html.gz (509 Bytes)
??:??:?? 0  /static/robots.txt.gz (42 Bytes)
??:??:?? 0  /static/site.webmanifest.gz (196 Bytes)
[   369][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=4 scl=15 freq=100000
??:??:?? 0 No I2C devices found
??:??:?? 0 Meshtastic hwvendor=2, swver=2.0.2.8146e84-d
??:??:?? 0 Setting random seed 1069679244
??:??:?? 0 Total heap: 201888
??:??:?? 0 Free heap: 168000
??:??:?? 0 Total PSRAM: 0
??:??:?? 0 Free PSRAM: 0
??:??:?? 0 NVS: UsedEntries 70, FreeEntries 560, AllEntries 630, NameSpaces 3
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 12
??:??:?? 0 OTA firmware version 0.1.1.98ab02e
??:??:?? 0 Initializing NodeDB
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved devicestate version 20
[   472][E][vfs_api.cpp:104] open(): /littlefs/prefs/config.proto does not exist, no permits for creation
??:??:?? 0 No /prefs/config.proto preferences found
??:??:?? 0 Installing default LocalConfig
??:??:?? 0 Setting default channel and radio preferences!
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 0, using UNSET
[   505][E][vfs_api.cpp:104] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
??:??:?? 0 No /prefs/module.proto preferences found
??:??:?? 0 Installing default ModuleConfig
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Loaded saved channelFile version 20
[   590][E][vfs_api.cpp:104] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
??:??:?? 0 No /oem/oem.proto preferences found
??:??:?? 0 Number of Device Reboots: 12
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 0, using UNSET
??:??:?? 0 region=0, NODENUM=0xf71e2cd4, dbsize=1
??:??:?? 0 Saving /prefs/db.proto
??:??:?? 0 No I2C devices found
??:??:?? 0 Read RTC time as 0
??:??:?? 0 Using MSL altitude model
??:??:?? 0 WANT GPS=1
??:??:?? 2 Module Info : 
??:??:?? 2 Soft version: .03 (45969)
??:??:?? 2 Hard version: 0040007
??:??:?? 2 Extensions:0
??:??:?? 2 UBlox GNSS init succeeded, using UBlox GNSS Module
??:??:?? 2 GxGSA NOT available
??:??:?? 2 External Notification Module Disabled
??:??:?? 2 Starting meshradio init...
??:??:?? 2 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
??:??:?? 2 Radio myRegion->freqStart / myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
??:??:?? 2 Radio myRegion->numChannels: 104
??:??:?? 2 Radio channel_num: 19
??:??:?? 2 Radio frequency: 906.875000
??:??:?? 2 Slot time: 42 msec
??:??:?? 2 Set radio: final power level=20
??:??:?? 2 Current limit set to 100.000000
??:??:?? 2 Current limit set result 0
??:??:?? 2 RF95 init result -19
??:??:?? 2 Frequency set to 906.875000
??:??:?? 2 Bandwidth set to 250.000000
??:??:?? 2 Power output set to 20
??:??:?? 2 Current limit set to 100.000000
??:??:?? 2 Current limit set result 0
??:??:?? 2 RF95 Radio init succeeded, using RF95 radio
??:??:?? 2 Not using WIFI
??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=237, time 3188 ms
??:??:?? 2 myNodeInfo.bitrate = 74.341286 bytes / sec
??:??:??���ݕ�M5�init, USB power=1
??:??:?? 2 Enter state: BOOT
[  2686][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
??:??:?? 2 [SerialModule] Serial Module Disabled
??:??:?? 2 [StoreForwardModule] Store & Forward Module - Disabled
??:??:?? 2 [RangeTestModule] Range Test Module - Disabled
??:??:?? 3 [GPS] publishing pos@0:2, hasVal=0, GPSlock=0
??:??:?? 3 [GPS] No GPS lock
??:??:?? 3 [GPS] onGPSChanged() pos@0, time=0, lat=0, lon=0, alt=0
??:??:?? 3 [GPS] updatePosition LOCAL pos@0, time=0, latI=0, lonI=0, alt=0
??:??:?? 3 [GPS] Node status update: 1 online, 1 total
??:??:?? 5 [PowerFSM] Enter state: POWER
??:??:?? 5 [PowerFSM] Initialise the NimBLE bluetooth module`

caveman99, 2022-11-15:
I'll try to dig up the schematics, but the board definition was not touched in ages. On some boards the oled must be powered specifically. Can you confirm the last working firmware version please?

hereiszee, 2022-11-15:
Thank you! Last working was 1.2.something.

Prior version didn’t need specific power I believe. I’ll also try booting an older version.

hereiszee, 2022-11-15:
In addition, board lights do not turn on.

caveman99, 2022-11-15:
Are you positively sure this is a version 1.0 board? 2 pushbuttons, antenna on a ipex pig tail attached right next to the oled?

[image]

caveman99, 2022-11-15:
Cause lights not turning on sounds like a wrong board definition.

hereiszee, 2022-11-15:
Correction, it is not a v1 board. 3 buttons, see attached photo. 

User error in that case? I used detect device to flash appropriate firmware. 
[image]

closed 2022-11-15 by hereiszee