meshtastic/firmware #1951: [Bug]: T-Beam GPS module not detected after low battery shutoff

xman5735, 2022-11-17:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.0.18ab874, 2.0.3.09fe616 Alpha

### Description

I have confirmed this with 4 of my ttgo t-beam v1.1. Low battery shutoff and occasionally hard resets will cause the device to output 'no gps module' and the log states 'GPS FactoryReset requested'. I then flash the device with the gps NMEA reset sketch modified by ErikThevking. GPS works again and then I reinstall the meshtastic software, everything works as expected.

I have the other logs when things are working properly and after the sketch, if wanted.

### Relevant log output

```Shell
??:??:?? 1 Battery: usbPower=1, isCharging=1, batMv=3546, batPct=9
??:??:?? 1 I2C device found at address 0x34
??:??:?? 1 axp192/axp2101 PMU found
??:??:?? 1 I2C device found at address 0x3c
??:??:?? 1 0x7 subtype probed in 2 tries 
??:??:?? 1 ssd1306 display found
??:??:?? 1 2 I2C devices found
??:??:?? 1 Read RTC time as 1
??:??:?? 1 Using MSL altitude model
??:??:?? 1 WANT GPS=1
??:??:?? 2 WARNING: Unable to disable NMEA GGL.
??:??:?? 2 GxGSA NOT available
??:??:?? 2 External Notification Module Disabled
??:??:?? 2 Turning on screen
??:??:?? 2 Module wants a UI Frame
??:??:?? 2 GPS FactoryReset requested
??:??:?? 3 Saving /prefs/db.proto
??:??:?? 3 Starting meshradio init...
??:??:?? 3 Set radio: region=US, name=LongFast, config=0, ch=19, power=30
??:??:?? 3 Radio myRegion->freqStart / myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
??:??:?? 3 Radio myRegion->numChannels: 104
??:??:?? 3 Radio channel_num: 19
??:??:?? 3 Radio frequency: 906.875000
??:??:?? 3 Slot time: 42 msec
??:??:?? 3 Set radio: final power level=20
??:??:?? 3 Current limit set to 100.000000
??:??:?? 3 Current limit set result 0
??:??:?? 3 RF95 init result -19
??:??:?? 3 Frequency set to 906.875000
??:??:?? 3 Bandwidth set to 250.000000
??:??:?? 3 Power output set to 20
??:??:?? 3 Current limit set to 100.000000
??:??:?? 3 Current limit set result 0
??:??:?? 3 RF95 Radio init succeeded, using RF95 radio
??:??:?? 3 Not using WIFI
??:??:?? 3 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=237, time 3188 ms
??:??:?? 3 myNodeInfo.bitrate = 74.341286 bytes / sec
??:??:?? 3 PowerFSM init, USB power=1
?�~??:?? 3 Enter state: BOOT
```

caveman99, 2022-11-17:
This is most likely a timing issue during boot. the 'no GPS' check is coming too early. You don't need to flash the GPS reset sketch by the way, the firmware is sending the same command sequence when it detects the chip needs to be reset.

xman5735, 2022-11-17:
Thank you. pyCLI claims GPS is enabled. I was unable to find a command to manually tell the board to re-check for gps module. So far, running that sketch is the only way I have found to get the GPS to work again. 

Considering the cause of the problem, could the esp32 be getting stuck in a low voltage situation and it causes the gps to disable? I know the esp8266 is susceptible to data corruption from that issue. 

Either way, if there is a way to reset it with a command instead of re-flashing, that is good enough for me.

xman5735, 2022-11-17:
Ill try and get my scope out and test the chip and the internal boost converter this weekend

rpsainio, 2022-12-15:
If the tbeam goes to deepsleep (SDS) and wakes up again then GPS is not powered up. It gets powered down just before deep sleep is initiated. I fixed this in my clone and wanted report as a separate issue, but then I found this one.

xman5735, 2022-12-16:
> If the tbeam goes to deepsleep (SDS) and wakes up again then GPS is not powered up. It gets powered down just before deep sleep is initiated. I fixed this in my clone and wanted report as a separate issue, but then I found this one.

I appreciate you figuring it out and posting about it. I've been focusing mostly on hardware so clearly I was wasting my time.

rpsainio, 2022-12-22:
The latest 2.0.8.090e166 Alpha looks like having the gps on/off fix for TBeam and at least switching gps works now and I will also test if it powers up GPS after  SDS

garthvh, 2023-01-26:
Let us know if you are still seeing this

closed 2023-01-26 by garthvh

rpsainio, 2023-01-26:
@garthvh  Thanks, I forgot to comment that GPS comes up properly after SDS