meshtastic/firmware #1952: [Bug]: CLI Configure command not working

tmarkson, 2022-11-17:
### Category

Serial

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.0.2.8146e84-d

### Description

Using the [example CLI command](https://meshtastic.org/docs/software/python/cli#--configure) produces an error, visible in log output section. 

### Relevant log output

```Shell
$ meshtastic --configure node.conf
Connected to radio
...
Aborting due to: Node.writeConfig() missing 1 required positional argument: 'config_name'
```

tmarkson, 2022-11-18:
Here is a [link to a thread](https://meshtastic.discourse.group/t/configuring-bluetooth-fixed-pin/6499) with the same symptoms

garthvh, 2022-11-18:
You need to use a yaml file

if you do meshtastic --export-config and save the output as a .yaml file you can re-import it using meshtastic --configure config.yaml