meshtastic/firmware #1953: [Feature request]: add ability for reboot noode by timer

D4rk4, 2022-11-19:
### Category

Hardware Compatibility, Other

### Hardware

Not Applicable

### Firmware Version

2.0.0.0

### Description

For autonomous nodes located in the middle of nowhere (like mountains) will be nice to have an automatic reboot feature by the timer. For example every 24h.

caveman99, 2022-11-19:
Is there a specific problem you want to tackle with this? Cause my nodes have been running rock solid for weeks, rebooting on the odd assert or memory allocation problem but they never hung. Ify they are really stuck, the HW watchdog kicks in and reboots them.

D4rk4, 2022-11-19:
Sometimes 2.0.x nodes based on TLORA stop transmitting anything to air, meshtastic --reboot, resolve this.
I'm sure it's not related to channel load because these nodes are located in different regions. And before the update to 2.0.x ran without any issues.
So to avoid this and potential future issues excellent to have a "maximum uptime" reboot trigger.

garthvh, 2022-11-19:
Duplicate of already closed issue https://github.com/meshtastic/firmware/issues/1193

closed 2022-11-19 by garthvh