meshtastic/firmware #1955: [Bug]: MQTT packet (?) crash firmware

D4rk4, 2022-11-19:
### Category

Other

### Hardware

T-Lora v1

### Firmware Version

2.0.3

### Description

Probably an MQTT-related defect.

`Module preferences: { "mqtt": { "enabled": true, "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats" },`

### Relevant log output

```Shell
20:44:08 2896 [Router] Incoming message was filtered 0xe6274a8
20:44:08 2896 [mqtt] Received MQTT topic msh/2/c/LongFast/!934cf8a8, len=88
20:44:08 2896 [mqtt] Using channel 0 (hash 0x8)
20:44:08 2896 [mqtt] Expanding short PSK #1
20:44:08 2896 [mqtt] Using AES128 key!
20:44:08 2896 [mqtt] ESP32 crypt fr=e6274a8, num=63bd7ca4, numBytes=28!
20:44:08 2896 [mqtt] decoded message (id=0x63bd7ca4 Fr0xa8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=67 rxSNR=11.75 rxRSSI=1.84105)
20:44:08 2896 [Router] Found existing packet record for fr=0xe6274a8,to=0xffffffff,id=0x63bd7ca4
20:44:08 2896 [Router] Resending implicit ack for a repeated floodmsg
20:44:08 2896 [Router] Should encrypt MQTT?: 0
20:44:08 2896 [Router] publish msh/2/c/LongFast/!25053ad4, 86 bytes
20:44:08 2896 [Router] Expanding short PSK #1
20:44:08 2896 [Router] Using AES128 key!
20:44:08 2896 [Router] ESP32 crypt fr=e6274a8, num=63bd7ca4, numBytes=28!
20:44:08 2896 [Router] enqueuing for send (id=0x63bd7ca4 Fr0xa8 To0xff, WantAck0, HopLim2 Ch0x8 encrypted rxSNR=11.75 rxRSSI=1.84105)
20:44:08 2896 [Router] txGood=595,rxGood=0,rxBad=0
20:44:08 2896 [Router] rx_snr found. hop_limit:2 rx_snr:11.750000
20:44:08 2896 [Router] rx_snr found in packet. Setting tx delay:2268
20:44:08 2896 [Router] Found existing packet record for fr=0xe6274a8,to=0xffffffff,id=0x63bd7ca4
20:44:08 2896 [Router] Add packet record (id=0x63bd7ca4 Fr0xa8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=67 rxSNR=11.75 rxRSSI=1.84105)
20:44:08 2896 [Router] Ignoring incoming msg, because we've already seen it (id=0x63bd7ca4 Fr0xa8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=67 rxSNR=11.75 rxRSSI=1.84105)
20:44:08 2896 [Router] Incoming message was filtered 0xe6274a8
20:44:08 2896 [mqtt] Received MQTT topic msh/2/c/LongFast/!25053a74, len=86

assert failed: T* MemoryDynamic<T>::alloc(TickType_t) [with T = _MeshPacket; TickType_t = unsigned int] MemoryPool.h:72 (p)

Backtrace:0x40084145:0x3ffd68900x40095f9d:0x3ffd68b0 0x4009bc59:0x3ffd68d0 0x400e278e:0x3ffd6a00 0x400df07f:0x3ffd6a20 0x400f1051:0x3ffd6a40 0x400f10c5:0x3ffd6af0 0x401f9b83:0x3ffd6b10 0x400fade9:0x3ffd6b30 0x400f0ac6:0x3ffd6b70 0x400d5086:0x3ffd6b90 0x400f5fbd:0x3ffd6bb0 0x400dd369:0x3ffd6be0 0x401231a9:0x3ffd6c00 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46
```

D4rk4, 2022-11-20:
It's reproducible:
```
09:10:24 4816 [mqtt] Received MQTT topic msh/2/c/LongFast/!25053ab4, len=86

assert failed: T* MemoryDynamic<T>::alloc(TickType_t) [with T = _MeshPacket; TickType_t = unsigned int] MemoryPool.h:72 (p)

Backtrace:0x40084145:0x3ffd68000x40095f9d:0x3ffd6820 0x4009bc59:0x3ffd6840 0x400e278e:0x3ffd6970 0x400df07f:0x3ffd6990 0x400f1051:0x3ffd69b0 0x400f10c5:0x3ffd6a60 0x401f9b83:0x3ffd6a80 0x400fade9:0x3ffd6aa0 0x400f0ac6:0x3ffd6ae0 0x400d5086:0x3ffd6b00 0x400f5fbd:0x3ffd6b20 0x400dd369:0x3ffd6b50 0x401231a9:0x3ffd6b70 

ELF file SHA256: 0000000000000000
```

caveman99, 2022-11-22:
this is the firmware running out of RAM. The json11 lib is a suspect in that one. i have made mods to exclude that more thoroughly in the latest alpha. Can you double check with 2.0.4 if this is still happening? (note, please explicitily DON'T enable the Json MQTT messages for that test)

caveman99, 2022-11-22:
possibly related to #1915

closed 2023-01-15 by garthvh