meshtastic/firmware #1961: [Security]:  Multiple security defects in C++ code in master branch

D4rk4, 2022-11-20:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

master

### Description

Hits = 226
Lines analyzed = 37778 in approximately 0.41 seconds (92739 lines/second)
Physical Source Lines of Code (SLOC) = 24212
Hits@level = [0]  54 [1]  44 [2] 149 [3]   7 [4]  26 [5]   0
Hits@level+ = [0+] 280 [1+] 226 [2+] 182 [3+]  33 [4+]  26 [5+]   0
Hits/KSLOC@level+ = [0+] 11.5645 [1+] 9.33421 [2+] 7.51693 [3+] 1.36296 [4+] 1.07385 [5+]   0
Dot directories skipped = 4 (--followdotdir overrides)
Minimum risk level = 1

### Relevant log output

```Shell
FINAL RESULTS:

./src/FSCommon.cpp:83:  [4] (buffer) strcpy:
  Does not check for buffer overflows when copying to destination [MS-banned]
  (CWE-120). Consider using snprintf, strcpy_s, or strlcpy (warning: strncpy
  easily misused).
./src/FSCommon.cpp:93:  [4] (buffer) strcpy:
  Does not check for buffer overflows when copying to destination [MS-banned]
  (CWE-120). Consider using snprintf, strcpy_s, or strlcpy (warning: strncpy
  easily misused).
./src/FSCommon.cpp:108:  [4] (buffer) strcpy:
  Does not check for buffer overflows when copying to destination [MS-banned]
  (CWE-120). Consider using snprintf, strcpy_s, or strlcpy (warning: strncpy
  easily misused).
./src/FSCommon.cpp:118:  [4] (buffer) strcpy:
  Does not check for buffer overflows when copying to destination [MS-banned]
  (CWE-120). Consider using snprintf, strcpy_s, or strlcpy (warning: strncpy
  easily misused).
./src/FSCommon.cpp:135:  [4] (buffer) strcpy:
  Does not check for buffer overflows when copying to destination [MS-banned]
  (CWE-120). Consider using snprintf, strcpy_s, or strlcpy (warning: strncpy
  easily misused).
./src/FSCommon.cpp:144:  [4] (buffer) strcpy:
  Does not check for buffer overflows when copying to destination [MS-banned]
  (CWE-120). Consider using snprintf, strcpy_s, or strlcpy (warning: strncpy
  easily misused).
./src/RedirectablePrint.cpp:41:  [4] (format) vprintf:
  If format strings can be influenced by an attacker, they can be exploited
  (CWE-134). Use a constant for the format specification.
./src/RedirectablePrint.cpp:47:  [4] (format) vsnprintf:
  If format strings can be influenced by an attacker, they can be exploited,
  and note that sprintf variations do not always \0-terminate (CWE-134). Use
  a constant for the format specification.
./src/RedirectablePrint.cpp:103:  [4] (format) vprintf:
  If format strings can be influenced by an attacker, they can be exploited
  (CWE-134). Use a constant for the format specification.
./src/RedirectablePrint.h:40:  [4] (format) printf:
  If format strings can be influenced by an attacker, they can be exploited
  (CWE-134). Use a constant for the format specification.
./src/RedirectablePrint.h:43:  [4] (format) vprintf:
  If format strings can be influenced by an attacker, they can be exploited
  (CWE-134). Use a constant for the format specification.
./src/SerialConsole.cpp:21:  [4] (format) vprintf:
  If format strings can be influenced by an attacker, they can be exploited
[... 813 lines cut ...]
./src/modules/CannedMessageModule.cpp:115:  [1] (buffer) strlen:
  Does not handle strings that are not \0-terminated; if given one it may
  perform an over-read (it could cause a crash if unprotected) (CWE-126).
./src/modules/CannedMessageModule.cpp:198:  [1] (buffer) strlen:
  Does not handle strings that are not \0-terminated; if given one it may
  perform an over-read (it could cause a crash if unprotected) (CWE-126).
./src/modules/CannedMessageModule.cpp:248:  [1] (buffer) strlen:
  Does not handle strings that are not \0-terminated; if given one it may
  perform an over-read (it could cause a crash if unprotected) (CWE-126).
./src/modules/ReplyModule.cpp:20:  [1] (buffer) strlen:
  Does not handle strings that are not \0-terminated; if given one it may
  perform an over-read (it could cause a crash if unprotected) (CWE-126).
./src/modules/SerialModule.cpp:231:  [1] (buffer) strlen:
  Does not handle strings that are not \0-terminated; if given one it may
  perform an over-read (it could cause a crash if unprotected) (CWE-126).
./src/modules/esp32/RangeTestModule.cpp:121:  [1] (buffer) strlen:
  Does not handle strings that are not \0-terminated; if given one it may
  perform an over-read (it could cause a crash if unprotected) (CWE-126).
./src/modules/esp32/StoreForwardModule.cpp:234:  [1] (buffer) strlen:
  Does not handle strings that are not \0-terminated; if given one it may
  perform an over-read (it could cause a crash if unprotected) (CWE-126).
./src/modules/esp32/StoreForwardModule.cpp:235:  [1] (buffer) strlen:
  Does not handle strings that are not \0-terminated; if given one it may
  perform an over-read (it could cause a crash if unprotected) (CWE-126).
```

MisterRager, 2023-03-11:
Could practically be renamed: `Use std::string`.

jp-bennett, 2023-08-04:
@D4rk4 What linter did you use to generate this list? Lots of code has changed since this was generated. I'd like to look over the current output, decide how many of them are problematic, and do a bit of cleanup work.

Djhg2000, 2023-11-06:
@jp-bennett I might be wrong but it looks like [Flawfinder](https://dwheeler.com/flawfinder/) to me.

garthvh, 2023-12-16:
This is no longer relevant, can be re-run.

closed 2023-12-16 by garthvh