meshtastic/firmware #1962: [Bug]: PowerFSM - Recording critical error 2

D4rk4, 2022-11-20:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.0.3

### Description

Device crash when no traffic is received.

`"power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 4294967295, "sdsSecs": 60, "lsSecs": 300, "minWakeSecs": 120 }`

### Relevant log output

```Shell
12:25:26 2039 [PowerFSM] radio wait to sleep, txEmpty=0
12:25:26 2039 [PowerFSM] radio wait to sleep, txEmpty=0
12:25:26 2039 [PowerFSM] radio wait to sleep, txEmpty=0
12:25:27 2039 [PowerFSM] radio wait to sleep, txEmpty=0
12:25:27 2039 [PowerFSM] radio wait to sleep, txEmpty=0
12:25:27 2040 [PowerFSM] radio wait to sleep, txEmpty=0
12:25:27 2040 [PowerFSM] NOTE! Recording critical error 2 at src/sleep.cpp:9d
```

thebentern, 2022-11-20:
Is there a reason why you have the super deep sleep interval as one minute? In my experience that's pretty aggressive for a device that consumes < 10ma idle. 
In the NRF52 devices, we go into super deep sleep after that timeout from our low power state, which in the case of these devices is screen off.
[image]
Having said that, there is definitely something fishy going on here. Is the device restarting after the critical error is recorded?

D4rk4, 2022-11-20:
@thebentern no, it's not rebooted after that.

garthvh, 2023-01-15:
NRF no longer deep sleeps

closed 2023-01-15 by garthvh