meshtastic/firmware #1963: [Bug]: Incorrect reporting about charging status on RAK4631

D4rk4, 2022-11-20:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.0.3

### Description

meshtastic not handle charging status/state of RAK4631 with RAK19007 base board.
In this case board connected to USB power and keep charging...

### Relevant log output

```Shell
13:21:21 542 [Power] Battery: usbPower=0, isCharging=0, batMv=4156, batPct=95
13:21:41 562 [Power] Battery: usbPower=0, isCharging=0, batMv=4163, batPct=96
13:22:01 582 [Power] Battery: usbPower=0, isCharging=0, batMv=4167, batPct=96
```

thebentern, 2022-11-20:
Detecting charging has been problematic on certain boards with simple charging ICs and VBatt measurement over ADC, particularly since the values tend to fluctuate on these boards. I've also found it to be dependent on the capacity / internal resistance of the cell.

D4rk4, 2022-12-02:
@thebentern Look like logic issue
When the charger was connected and battery charged meshtastic reported something like:
`[Power] Battery: usbPower=1, isCharging=1, batMv=4256`

garthvh, 2022-12-03:
You can try setting the ADC multiplier override value

closed 2022-12-03 by garthvh