meshtastic/firmware #1966: Feature request: fix broken external notification module

tropho23, 2022-11-21:
The external notification module does not function in the 2.x firmware; I have tested nrf52 and ESP32 boards, and have confirmed with my multimeter that no voltage is passed (thus no notification triggered) for any pin when the external notification plugin is enabled.  I've tried setting multiple pins as outputs, including ones that I believe previously worked with 1.3.x firmware and no notifications are possible.  meshtastic --info confirms the module is enabled, notifications for messages and bell are enabled, and GPIO is set with a duration in seconds.

Andrey-VD, 2022-11-23:
I connected the LED to pin 13. Notifications  work only for  messages over GPIO channel.
 Firmware v 2.0.2

caveman99, 2022-11-23:
This is fixed in develop branch and will be part of next firmware.

closed 2022-11-23 by caveman99