meshtastic/firmware #1969: [Feature Request]: manual buzzer option for external notification plugin

BeenThereScrewedUp, 2022-11-21:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.2.8146e84

### Description

For some boards, PIN_BUZZER is defined in the respective variant.h file.
For the T-Beam board, no buzzer is defined - which is correct as it does not feature an on board buzzer.

I connected a buzzer to pin 13 and GND and configured meshtastic accordingly. The debug log activates pin 13 as a digital pin instead of as a pwm pin. This seems to be the case as no PIN_BUZZER is defined for the board. However, there is no option to define the pin as pwm pin without recompiling the firmware with an modified variant.h.

Would it be possible to allow a manual flag to be set in the notification configuration to choose between digital and pwm output?

If not it would be nice if you could someone could update the documentation to indicate that a buzzer relying on tone.h and pwm is only possible on boards with a built-in buzzer and otherwise only digital out is available.

garthvh, 2022-11-22:
Detected PWM modules should be used for ease of use, GPIO should be the manual option as it is much more finicky for regular users.

garthvh, 2023-01-15:
This is in place now settings are on device and external notification

closed 2023-01-15 by garthvh