meshtastic/firmware #1981: [Bug]:

IhorNehrutsa, 2022-11-24:
### Category

BLE

### Hardware

Other

### Firmware Version

2.0.5

### Description

I try to connect from Android Meshtastic app to DIY module based on ESP32 and EBYTE E22 LoRa module without the screen.
Android app offers use PIN code 0000 or 1234
[image]
But default BLE pin is 123456
https://github.com/meshtastic/firmware/blob/c87cd136d4398d69829327e28145ee20a845f43d/src/DebugConfiguration.h#L38

It may also be related to the NimBLE-Arduino library
https://github.com/h2zero/NimBLE-Arduino/blob/0d9f039ba7f960946a8c9aeb639c3a64294f0d60/src/NimBLEDevice.cpp#L69
https://github.com/h2zero/NimBLE-Arduino/blob/0d9f039ba7f960946a8c9aeb639c3a64294f0d60/src/NimBLEClient.cpp#L1278-L1281
https://github.com/h2zero/NimBLE-Arduino/blob/0d9f039ba7f960946a8c9aeb639c3a64294f0d60/src/NimBLEServer.cpp#L899-L902

caveman99, 2022-11-24:
That's your phone suggesting a 'typical' pairing pin. That message can not be influenced by Meshtastic as it is hardwired in the Android OS.

closed 2022-11-24 by caveman99

IhorNehrutsa, 2022-11-24:
Perhaps it is better to use 1234 as default PIN code?
 #define defaultBLEPin 1234

caveman99, 2022-11-24:
BLE Pin needs to be 6 characters. Modern phones won't accept 4 for pairing any more.

IhorNehrutsa, 2022-11-25:
Then we need proper documentation for the default PIN of 123456. (for nodes without screen).

garthvh, 2022-11-25:
https://meshtastic.org/docs/settings/config/bluetooth

IhorNehrutsa, 2022-11-25:
@garthvh @caveman99 Thanks.