meshtastic/firmware #1983: Feature request: Update Store and Forward module for 2.x firmware

tropho23, 2022-11-24:
Request resurrection of the Store and Forward (S&F) module for 2.x firmware.  Previous work stalled on this module for unknown reasons (time, complexity?) and it is a great capability for any mesh.

The primary use case is for any device to receive and store previously sent (but missed) messages from a S&F 'server' node (router role perhaps?) with onboard PSRAM to enable a much larger message repository, if needed.  The PSRAM-capable device would likely need to be ESP32-based as nrf52-based devices lack PSRAM.

The secondary use case is to use a device's stored message repository (via enabled S&F module) on standalone messenger devices with either keyboard or rotary encoder as well.  This would allow more than a single message to be displayed on the device screen, which is the current situation; as of today users may only view the last received message, which is lost when another arrives.

garthvh, 2023-12-16:
Need to add tapback and replies, and look at if it is feasible to do position packets 

https://github.com/meshtastic/firmware/issues/1042

https://github.com/meshtastic/firmware/issues/2399

garthvh, 2023-12-16:
@caveman99 I think a bunch of this is done.  I still am not seeing any actual messages when I request a client_history?  Do the store and forward packets just provide the data and the messages just come back as regular messages?

garthvh, 2023-12-21:
This is done, there are likely some bugs

closed 2023-12-21 by garthvh