meshtastic/firmware #1984: Feature request: Continuous GPIO-enabled activity until user intervenes

tropho23, 2022-11-24:
Request capability to enable continuous/periodic GPIO-enabled activity (light/flash LED, piezo buzzer chirping, etc.) until user intervenes with a button or keypress.  Ideally would be tied to user button/IO5 so nearly all devices can benefit, but configurable for devices without IO5 or DIY boards that need to use a different one.

Example use case: user sets 'continuous_notification' setting to 'true', configures 'continuous_notification_period' to '120', and configures 'continuous_notification_pin' to '10'.  In this example, when the device receives a message it would keep a LED connected to pin 10 lit for the duration of time specified in the current GPIO output duration setting, reactivate the LED every 120 seconds, then stop this continuous reactivation when the user presses IO5.

With this capability enabled and configured we would have a means to keep notifying the user until the user acknowledges and dismisses the notification.

caveman99, 2022-12-08:
https://github.com/meshtastic/firmware/commit/32d9a397aa5ef0515fd6344bcd17cb2b4f7015bc covers this. Please test so it can be included in next release

garthvh, 2023-01-26:
@tropho23 is this working?

tropho23, 2023-01-26:
Yes, thank you for following up.  I meant to request closure of this ticket after caveman99 implemented this feature; I've tested it and it works perfectly.

Tony

closed 2023-01-26 by garthvh