meshtastic/firmware #1988: [Bug]: T-Beam  "No GPS Lock"

ReiHof, 2022-11-25:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.4.5417671  and  2.0.5.65e8209

### Description

Since V2.0.4  T-Beam shows  "No GPS Lock".  Red led blinks every second. 

V2.0.3 shows  valid GPS positions 

### Relevant log output

```Shell
16:36:23 52 [GPS] SOME data is TOO OLD: LOC 105, TIME 105, DATE 106
16:36:23 53 [GPS] SOME data is TOO OLD: LOC 207, TIME 207, DATE 208
16:36:23 53 [GPS] SOME data is TOO OLD: LOC 309, TIME 309, DATE 310
16:36:23 53 [GPS] SOME data is TOO OLD: LOC 411, TIME 411, DATE 412
16:36:23 53 [GPS] SOME data is TOO OLD: LOC 513, TIME 513, DATE 514
16:36:23 53 [GPS] SOME data is TOO OLD: LOC 615, TIME 615, DATE 616
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
16:36:23 53 [WebServerThread] webAPI handleAPIv1FromRadio
16:36:23 53 [WebServerThread] webAPI handleAPIv1FromRadio, len 0
16:36:23 53 [GPS] SOME data is TOO OLD: LOC 716, TIME 716, DATE 717
16:36:23 53 [GPS] SOME data is TOO OLD: LOC 817, TIME 817, DATE 818
16:36:23 53 [GPS] SOME data is TOO OLD: LOC 919, TIME 919, DATE 920
16:36:24 53 [GPS] SOME data is TOO OLD: LOC 2, TIME 2, DATE 3
16:36:24 53 [GPS] SOME data is TOO OLD: LOC 105, TIME 105, DATE 106
16:36:24 54 [GPS] SOME data is TOO OLD: LOC 206, TIME 206, DATE 207
16:36:24 54 [GPS] SOME data is TOO OLD: LOC 309, TIME 309, DATE 310
16:36:24 54 [GPS] SOME data is TOO OLD: LOC 411, TIME 411, DATE 412
16:36:24 54 [GPS] SOME data is TOO OLD: LOC 513, TIME 513, DATE 514
16:36:24 54 [GPS] SOME data is TOO OLD: LOC 615, TIME 615, DATE 616
16:36:24 54 [GPS] SOME data is TOO OLD: LOC 717, TIME 717, DATE 718
16:36:24 54 [GPS] SOME data is TOO OLD: LOC 819, TIME 819, DATE 820
16:36:24 54 [GPS] SOME data is TOO OLD: LOC 3, TIME 3, DATE 3
16:36:25 54 [GPS] SOME data is TOO OLD: LOC 104, TIME 104, DATE 105
```

caveman99, 2022-11-25:
this is a known issue and will be fixed in 2.0.6

garthvh, 2022-11-30:
Please test with 2.0.6, you may have to power cycle twice (with the power button not reset) waiting 60 seconds between.

closed 2022-11-30 by garthvh