meshtastic/firmware #1989: [Feature Request]: Add Device.Role in Metadata

garthvh, 2022-11-25:
### Platform

NRF52, ESP32

### Description

It would be helpful to know what the device role is when requesting metadata from a remote node so that messaging can only show clients if a network has a lot of ROUTER devices that are not connected to apps.

caveman99, 2022-12-30:
Are you talking about the periodic NodeInfo broadcasts or a certain admin packet you have to requests verbatim?

garthvh, 2022-12-30:
> Are you talking about the periodic NodeInfo broadcasts or a certain admin packet you have to requests verbatim?

This message https://github.com/meshtastic/protobufs/blob/e406029f68d8a554f43fd1f250baca6b3c1a9a33/device_metadata.proto#L13

closed 2023-01-25 by garthvh