meshtastic/firmware #1995: [Bug]: TBeam Router random freeze to logo splash screen.

chopstar001, 2022-11-26:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.3.09fe616

### Description

Recently updated all my T-Beams from 1.2.65 to the latest Beta 2.0.3. 
I have 13 T-Beams That are set to ROUTER. 5 of the 13 Routers had a recurring random freeze to the splash screen. This happens after the device has been running for a number of hours. I would have to press the reset button,then they would be fine for a few more hrs, sometimes a day. But then they would freeze to splash screen again.
Note, if device is left for to long in this frozen state, I have found it eventually kills the battery. I now have four dead Panasonic batteries from this, one of them actually exploded.
So a work around for me on the problem devices was to set them as ROUTER_CLIENT, and the Random freeze event stopped.
The only power settings changed was Super Deep Sleep Duration, from 86400 to 9000 and then I had to bump ADC Multiplier Override ratio up to 2 in order to save the new setting.
[image]

garthvh, 2022-11-26:
The ADC multiple override is completely ignored on a tbeam, can you post your meshtastic --info

What are you trying to do by reducing the super deep sleep value?

chopstar001, 2022-11-28:
[esp-web-tools-logs.txt](https://github.com/meshtastic/firmware/files/10099398/esp-web-tools-logs.txt)
So I turned on logging, waited for the device to repeat the problem. Not sure if I'm extracting the log file the correct way though. I used web tools via serial. and it looks like its only capturing post initial error, and as the act of plugging into serial caused a state change, leading to what looks like a boot loop for about a minute or so.
[esp-web-tools-logs (1).txt](https://github.com/meshtastic/firmware/files/10099513/esp-web-tools-logs.1.txt)

I have been using a reduced super deep sleep value on all my solar powered routers for about 4 months now to alleviate the issue of the nodes not waking up after experiencing consecutive days of no sun. Default value was a year. Now, essentially they just boot up as soon as they have enough charge.

garthvh, 2022-11-29:
Running batteries flat is a really bad idea, if you are modifying the SDS value you need to let it go at least a day, your battery exploded because of your two hour SDS value which does not give the unit enough time to get a reasonable charge after it loses power.

garthvh, 2022-11-29:
> [esp-web-tools-logs.txt](https://github.com/meshtastic/firmware/files/10099398/esp-web-tools-logs.txt) So I turned on logging, waited for the device to repeat the problem. Not sure if I'm extracting [...]

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x40081ade  PS      : 0x00060035  A0      : 0x40086af4  A1      : 0x3ffbf92c  
A2      : 0x00000000  A3      : 0x3ffbe3d4  A4      : 0x80084684  A5      : 0x4009727e  
A6      : 0x3ffbd2b0  A7      : 0x00000804  A8      : 0x00000001  A9      : 0x3ffbf90c  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffba4d0  
A14     : 0x3ffbd2b0  A15     : 0x0000000c  SAR     : 0x00000006  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40092a50  LEND    : 0x40092a5b  LCOUNT  : 0xffffffff  
Core  1 was running in ISR context:
EPC1    : 0x40139b3f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x40081adb:0x3ffbf92c |<-CORRUPTED

Core  0 register dump:
PC      : 0x40207c92  PS      : 0x00060835  A0      : 0x801384f0  A1      : 0x3ffbd6c0  
A2      : 0x00000000  A3      : 0x4009200c  A4      : 0x00060820  A5      : 0x3ffbcbc0  
A6      : 0x007bf978  A7      : 0x003fffff  A8      : 0x8013802e  A9      : 0x3ffbd690  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x8009ab24  A13     : 0x3ffbd5c0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x40207c8f:0x3ffbd6c00x401384ed:0x3ffbd6e0 0x400987b0:0x3ffbd700

garthvh, 2022-11-29:
Please post the details from either meshtastic --info or meshtastic --export-config

garthvh, 2022-11-29:
Are you using wifi on a solar powered repeater? It looks like it from the backtrace. That is never going to work with that panel small panel and an ESP32 using 200mah with wifi on.

chopstar001, 2022-11-30:
With good sun battery is usually Fully charged by end of day, they are 800ma panels. 

Re wifi, I temporally turned it on to see if I could extract the log file over http.

The battery that blew up, pretty sure was a new Panasonic. I'm up to 6 dead batteries now so I know at least one or two of those were new batteries. And this all in the space of a few days after updating to 2.0.3, from the 5 specific routers in question. Interesting its only happening with only these and not the others with same configs...some are SX1262's some just LORA32. I'm thinking it may just be the hardware is damaged. I dont Know. I had been using that low SDS value for months without a hitch at all.

[Info.txt](https://github.com/meshtastic/firmware/files/10118247/Info.txt)

Note: In this info youll see I have updated to the latest firmware 2.0.6
I've taken your advice on the SDS. it's now back to its default value. See how she goes....or blows. Thanks.

chopstar001, 2022-11-30:
Just an update, after update to 2.0.6 and SDS to default value. Still froze to logo, after around 7rs of running time.

garthvh, 2022-11-30:
The device is crashing, we need to see the config values, the back trace shows a crash from Wi-Fi and GPIO

chopstar001, 2022-11-30:
user_prefs:
bluetooth:
enabled: true
fixedPin: 123456
device:
role: ROUTER
serialEnabled: true
display:
autoScreenCarouselSecs: 6
screenOnSecs: 600
lora:
hopLimit: 6
region: ANZ
txEnabled: true
txPower: 30
usePreset: true
network:
ntpServer: 0.pool.ntp.org
position:
fixedPosition: true
gpsAttemptTime: 300
gpsUpdateInterval: 86400
positionBroadcastSecs: 10800
positionFlags: 163
power:
adcMultiplierOverride: 2.0
lsSecs: 86400
meshSdsTimeoutSecs: 4294967295
minWakeSecs: 10
sdsSecs: 86400
waitBluetoothSecs: 1

chopstar001, 2022-11-30:
> The device is crashing, we need to see the config values, the back trace shows a crash from Wi-Fi and GPIO

user_prefs:
bluetooth:
enabled: true
fixedPin: 123456
device:
role: ROUTER
serialEnabled: true
display:
autoScreenCarouselSecs: 6
screenOnSecs: 600
lora:
hopLimit: 6
region: ANZ
txEnabled: true
txPower: 30
usePreset: true
network:
ntpServer: 0.pool.ntp.org
position:
fixedPosition: true
gpsAttemptTime: 300
gpsUpdateInterval: 86400
positionBroadcastSecs: 10800
positionFlags: 163
power:
adcMultiplierOverride: 2.0
lsSecs: 86400
meshSdsTimeoutSecs: 4294967295
minWakeSecs: 10
sdsSecs: 86400
waitBluetoothSecs: 1

garthvh, 2022-11-30:
Be very careful with hops, that is likely your biggest problem, would probably be good for you to run the meshtastic simulator. There is generally very little message delivery benefit to increasing hops, and it can substantially increase network traffic and power usage.

I would also not set the carousel to every 6 seconds on an unattended node solar powered node, in fact the carousel is not really a good idea on router node.

Monitor the channel utilization of your nodes, should not be pegged at 25% as I suspect these nodes are. High channel utilization is going to affect routers even more and draw a lot of power.

closed 2022-11-30 by garthvh