meshtastic/firmware #2003: [Feature Request]: Information which devices are within range of my antenna and which we can see through the mesh.

EmilM256, 2022-11-28:
### Platform

ESP32

### Description

Sometimes it is not known which nodes I have direct radio communication with and which node sends information via mesh. This is useful information, especially when we are examining the network coverage, and we use several Routers.
Is it possible to add a marker (letters, icons?) on the OLED screen and in the application in the second tab with the list of users?
It would also be nice to display the "hop limit" of received messages in the chat. For example, a date wheel and a cloud.

garthvh, 2022-11-28:
There is no trace route like functionality with flood routing, you can read some more details on the mesh here https://meshtastic.org/docs/overview/mesh-algo

EmilM256, 2022-11-28:
I do not mean the traceroute function as in IP4. Information only, packet routed or not (ie direct). Modem lora probably knows from which device the packet got (physically). I would like to know which nodes it contacts directly - the rest are routed - no matter what way.

EmilM256, 2022-11-28:
Ok, if the packet comes without any MAC address or other tag from which device it was sent, I understand.

Option two, information about Hop Limit? If I get a telemetry package with hop limit 3 (standard settings) I know that 3 means a direct connection. Less than 3, routed connection. The same can be shown in the chat?
I'm right? could this be added?

caveman99, 2022-11-28:
we don't know the start vale of the hop counter, this can be configured per device.

garthvh, 2022-11-28:
Take a look at the simulator if you are wanting to do network planning https://github.com/GUVWAF/Meshtasticator

closed 2022-11-28 by garthvh

EmilM256, 2022-11-29:
caveman99 - yes, the hop limit value can be changed in the configuration, but it doesn't prevent you from simply displaying it.
For me, a simple view like below would be useful.
[image]
(sorry for the ugly graphics)