meshtastic/firmware #2005: Heltec display not honoring screen timeout

ghostop14, 2022-11-28:
The display on the heltec v2.1 is not turning off after the "Screen on for" setting. 

Test setup: No other mesh devices on, running 2.0.6 firmware on heltec v2.1, device paired via bluetooth to mobile app on iphone.

I tried disabling the bluetooth to see if it was related to being paired, but seemed to ignore the setting whether paired or unpaired.

ghostop14, 2022-11-28:
@rcarteraz opened issues as I was running into them.  Missed your comment on 2004.  Odd on the heltec support.  It has been there for some time and the default firmware packages still build the v2 and v2.1 versions.  I did see v3 get added but only with v2.0.3, but the other versions are still bundled in the firmware packages.  Seems like the hardware list on the website may be shrinking rather than expanding?  I'll see if I can get my hands on some v3 boards.

caveman99, 2022-11-29:
the older heltec boards have bad thermal issues and suffer from bad circuit layout and component sizing. I did the adaptation to the really old V1 board and believe me they're messed up! Yes, they are somewhat supported so we keep building firmware images for the existing devices. New buyers are strongly advised to use V3 or the Lora Stick as many problems with previous designs seem to be addressed in them. Support status is about the same level as the ancient T-Beam V0.7 and the below-V2.1 TLora Boards.

closed 2022-11-29 by caveman99

ghostop14, 2022-11-29:
Thanks @caveman99 great background.  I have some v3's on order.  The boards were the perfect fit for a current design, so fingers crossed. I'm definitely seeing some oddities with even the v2's hopefully they've fixed in v3.

MotoG7rooted, 2022-12-30:
I just found out if you have low power mode on in the device settings for the v2 board the display does actually honor the display timeout settings. Hope that helps!