meshtastic/firmware #2006: [Bug]: T-Beam display not honoring screen timeout

Anonym-tsk, 2022-11-29:
### Category

Hardware Compatibility, Other

### Hardware

T-Beam

### Firmware Version

2.0.6.97fd5cf

### Description

The display on the T-Beam is not turning off after the "Screen on for" setting.

Test setup: 2.0.6 firmware on T-Beam, WiFi connection.

I tried screen on time from 30 to 600 seconds.

caveman99, 2022-11-29:
if the device is powered by USB, the screen will not turn off.

Anonym-tsk, 2022-11-29:
Why? I worry about oled screen burn-in.

garthvh, 2022-11-29:
> Why? I worry about oled screen burn-in.

Not a real big concern https://hackaday.com/2019/04/23/a-year-long-experiment-in-oled-burn-in/

Anonym-tsk, 2022-11-29:
> Not a real big concern

I have diy co2 sensor with oled display. Some pixels burned-in after 2 years.

garthvh, 2022-11-29:
> > Not a real big concern
> 
> I have diy co2 sensor with oled display. Some pixels burned-in after 2 years.

Of being on the same screen, which is not an issue for meshtastic

closed 2022-11-29 by garthvh

Anonym-tsk, 2022-11-29:
Would you make configuration parameter "Turn off screen also if device is powered"?