meshtastic/firmware #2008: [Bug]: T-ECHO GPS: 'no GPS module found'

phxyz12, 2022-11-29:
### Category

Hardware Compatibility, Other

### Hardware

T-Echo

### Firmware Version

2.0.6.97fd5cf.uf2

### Description

Hi all

I got two brand new Lilygo TTGO Meshtastic T-Echo 868 MHz BME280 devices. When trying to couple them with the Android app, I got a message saying the the device's firmware is too old to work with the app. Following the instructions on the Meshtastic website, I reased the devices and installed latest firmware 2.0.6.97fd5cf.uf2.

On both devices I now see the message 'no GPS module found' and GPS is not working. I can't say if this was different with the factory-installed firmware version of Meshtastic.

The situation is the same on both devices.

caveman99, 2022-11-29:
Do you know what version was installed before you updated? 1.3.10 perchance?

phxyz12, 2022-11-29:
> Do you know what version was installed before you updated? 1.3.10 perchance?

Thanks for your response. Sadly, I do not know the exact version. But I remember it starting with 1.3

phxyz12, 2022-11-30:
I made some tests:

Firmware 2.0.6 - 'no GPS module', 'no GPS'
Firmware 2.0.3 - 'no GPS module', 'no GPS'
Firmware 2.0.0 - 'no GPS module', 'no GPS'
Firmware 1.2.65 - 'no GPS Lock', 'no Sats'

It looks as if it works with FW 1.2.65, but is already broken in FW 2.0.0 and all following FW versions.

Sadly, firmware 1.2.65 is not compatible with the Android app anymore, so I cannot use it.

caveman99, 2022-11-30:
we had a few firmware versions where the baudrate of the GPS was reduced to 4800. we had a bit of code in the firmware up to 1.3.42 to rectify that and set it back to 9600. I am running 2.0.4 at the moment and it's showing 'no Sats' for me, so the GPS is communicating. Not sure if there's something similar to the T-Beam GPS reset firmware that factory-resets the L76K Chip inside the T-Echo. The (assumed) 1.3.10 firmware has exactly that speed reduction, so i think this is what is biting you.

phxyz12, 2022-11-30:
Results:

I flashed both T-Echo devices again with firmware 2.0.6, both work with GPS. I do not safely know if this will survive a real power-off e.g. by an empty battery. Nevertheless, I consider this issue as solved. In case of any troubles reoccurring, I would reopen it.

Thanks a lot, caveman99

closed 2022-11-30 by thebentern

UberTesla, 2022-12-20:
I have this exact same issue. GPS works with T-Echo_Meshtastic-1.2.64.fc48fcd.uf2 downloaded via LilyGo's Repo at https://github.com/Xinyuan-LilyGO/T-Echo/tree/main/examples/Meshtastic_firmware

When I try to upgrade to 2.0.6.97fd5cf or 2.0.7.91ff7b9 I get No GPS Module.

CodeForMeThal, 2022-12-24:
Same issue for me. I don't have access to version 1.2.x. Thus, I cannot try the described solution. But using v 1.3.x instead, and then upgrade to 2.0.6 gps does not work. But 1pps-LED is blinking each second. I guess NMEA is the root cause because this is what has been changed at 2.0.0!

UberTesla, 2022-12-25:
> Same issue for me. I don't have access to version 1.2.x. Thus, I cannot try the described solution. But using v 1.3.x instead, and then upgrade to 2.0.6 gps does not work. But 1pps-LED is blinking each second. I guess NMEA is the root cause because this is what has been changed at 2.0.0!

You can find version 1.2 in lilygo's repo. This is the manufacturer of the board.

https://github.com/Xinyuan-LilyGO/T-Echo/tree/main/examples/Meshtastic_firmware

That said, upgrading from 1.2 to 2.0.6 or 1.2 to 2.0.7 did not fix the issue for me.

garthvh, 2022-12-25:
Please post an issue on the techo GitHub, we have no source available for the 1.2.64 release on their site, so what it does differently from the main firmware is a mystery to the meshtastic team. There does not seem to be a GPS reset sketch available for the techo but flashing softrf or their gps.ino sketch and back may help.

CodeForMeThal, 2022-12-25:
Hi,Thanks a lot for that hint - I will try it soon.Happy Xmas!

   
   

   
  https://github.com/Xinyuan-LilyGO/T-Echo/tree/main/examples/Meshtastic_firmware

guibom, 2023-04-02:
I'm also having this issue with a new T-Echo...  
It detects the GPS module with the linked firmware 1.2.64. However it does not work with any newer firmware, like 2.1.5.

As a test, I've also flashed the GPS.ino sketch from TTGO and that worked fine. 
And that's just a basic sketch that connects to the GPS with 9600 baud, and doesn't do anything special. 

Here are the things I've tried:

- Use reset firmware, then install 1.2.64. Then install newer one on top (both with and without resetting it first).
- Using the GPS.ino sketch, tried to send the same factory reset command form Meshtastic `NMEAGPS.cpp`.  Then installed Meshtastic. That didn't seem to do anything.
- Tweaked a bit the Meshtastic firmware to change the GPS initialization so it more closely matches the TTGO gps example. Also didn't work, but probably because of my limited knowledge here.

I believe this bug needs to be re-opened. How were you able to make it work @phxyz12 ??

garthvh, 2023-04-02:
The techo does not use a ublox gps, so not sure what the ublox arduino code may have done

guibom, 2023-04-02:
> The techo does not use a ublox gps, so not sure what the ublox arduino code may have done

Sorry, to be clear I'm talking about the specific GPS example from TTGO for the T-Echo... It's not a ublox one.
This one here: https://github.com/Xinyuan-LilyGO/T-Echo/blob/main/examples/GPS/GPS.ino

Jeripimo, 2023-08-21:
> 

Hi, having simular problem.

What firmware did you use to set the gps baudrate to 9600? Would you be so kind to share it so I can try.

thanks, Jerome

> caveman99

macca100, 2024-03-15:
I had the same issues as other people in this thread when I updated the firmware on my T-Echo. I solved the GPS issue by doing the following:

Radio configuration -> Position
GPS Mode: Enabled
Redefine GPS_RX_PIN: 9
Redefine GPS_TX_PIN: 8
Redefine PIN_GPS_EN: 2

UberTesla, 2024-03-15:
> I had the same issues as other people in this thread when I updated the firmware on my T-Echo. I solved the GPS issue by doing the following:
> 
> Radio configuration -> Position GPS Mode: Enabled Redefine GPS_RX_PIN: 9 Redefine GPS_TX_PIN: 8 Redefine PIN_GPS_EN: 2

You're a legend. And I just realized it was on the back of the unit this whole time!