meshtastic/firmware #2024: [Bug]:  can't configure node on Android after having paired with an iOS client.

dikkedimi, 2022-12-05:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.7

### Description

After pairing one of my radio's (e188) with iOS I am having trouble connecting it to my other smartphone, an android device. Even though I told my iOS device to forget the pairing credentials. This unit paired with the android phone without any problems prior to connecting it to the iOS device.

I am able to connect using the "+" button in the android app. But after pairing, the app won't let me change radio settings. It also won't connect to the channel that is configured by the iOS app. Further more, both radio's then had the same name, so I'm guessing iOS sets a few config options that the android app does not overwrite?

I am able to send a message directly to the other node. It's being acknowledged by the other radio, but I don't receive the message on my android smartphone. The message is visible on the sending node's display though, but not on the receiving node's display.

On a Previous firmware version (can't remember which) this same radio borked after pairing with iOS for some reason, and got stuck in a boot loop. I also was not able to flash the device from my imac any longer. I've resolved the latter issue by removing and reinstalling the CH340 driver (had to disable SIP to get the kext to be loaded). Put on the new 2.0.7. firmware and now I have this current issue.

Unfortunately I am not getting any debug messages on the android device, so it appears the communication between the Android phone and the e188 node is not actually being utilized. Nodes are communicating with eachother (so radio settings match those set by the iOS app). But as mentioned I am unable to get the messages to appear on my android phone and I am not able to adjust radio settings in the app.

caveman99, 2022-12-05:
are you trying to connect one radio to two phones at the same time? Because that will not work.

dikkedimi, 2022-12-05:
> are you trying to connect one radio to two phones at the same time? Because that will not work.

to clarify: no.

Like mentioned, I unpaired the radio from my iOS phone. but it seems the (radio) config put in by the iOS app is pervasive. the android app apparently can't modify the config or communicate with the node. Pairing works, but am not able to adjust the radio settings and receive/send messages through the node from Android.

dikkedimi, 2022-12-05:
running version 1.3.39 on android.

Under the map tab, I am not able to enable 'provide location to mesh', just like I am unable to modify the radio settings in the android app.

caveman99, 2022-12-05:
yes, because 1.3.x is not supported any more. You need to get a 2.0.x Android app, firmware and iOS app.

closed 2022-12-05 by dikkedimi