meshtastic/firmware #2032: [Feature Request]: Clone watchdog/LED1 output to any GPIO pin

D4rk4, 2022-12-09:
### Platform

NRF52, ESP32

### Description

Please add the ability to mirror the heartbeat "blinking" signals to any user-configurable GPIO pin.
This feature is required to efficiently integrate "hardware" external watchdogs into autonomous Meshtastic nodes/repeaters located in hard accessible places.

garthvh, 2022-12-09:
This is already in place on most of the boards without adding any configuration, what exactly are you trying to do with which board?

D4rk4, 2022-12-10:
For example RAK19007

garthvh, 2022-12-10:
There are already pins on the 19007 you can use, check the data sheet for led2, led1 is the charger I think and does not get managed from the firmware.

garthvh, 2022-12-10:
> @garthvh so, how I can pick up the GREEN LED heartbeat signal on this board?

Check the data sheet, RAK offers extensive documentation

garthvh, 2022-12-11:
There is a pin for led2, which is what you want, not sure what 5d is

closed 2022-12-11 by garthvh

garthvh, 2022-12-11:
There are no usable GPIO pins available on the base board, you need an IO board. 

There is nothing to do with this request.