meshtastic/firmware #2038: [Feature Request]: Request battery status

SR7673, 2022-12-13:
### Platform

ESP32

### Description

When i connect to my mesh (i.e. after a reebot of my local node), the distant repeater tells me its location. Fine. In the Android app i can select "direct message" or "request position". That's fine! I would like to be able to request the battery status as well.
Currently it seems not to be transferred automatically over here.

Thanks for the great project again!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! 
Regards, Stefan

SR7673, 2022-12-13:
That looks good, on your side. Here i'm not seeing this battery information. I guess it depends on the role of a node. My local node is a client (of course) and the other one is in repeater mode (battery saving, no display and so on).

I remember that i also saw battery informations, just like you. Don't know, maybe as you say, a periodic transfer of that information. However it would be good to be able to request that info, on demand, not (only) periodically.

In fact it can be useful not to broadcast the battery status periodically at all because this causes more channel usage and, in summer, it is uncritical anyway. In winter (we have -5 °C now here) it is more economic just to get this information on request, also to save energy...

garthvh, 2022-12-13:
Does the router have the same primary channel key?

SR7673, 2022-12-13:
Yes, it has the same channel key (if the key would be different, i wound't "see" it at all).
The router is mounted on the top of a high building, 38m above ground. Re-transmitting messages is the reason why it is there, it is its task.

SR7673, 2022-12-13:
BTW the client uses V2.06, the router/repeater uses V2.06

garthvh, 2022-12-13:
The router should be sending its device telemetry every 15 minutes by default if you are not getting those packets, requesting the same packet is probably not going to work either. The easiest option may be to make device telemetry broadcasts less frequent.

The request position is a little bit misleading as far as what it is doing behind the scenes, it is sending your position to the node and then requesting a response back, device telemetry works the same way I believe.

This should already be possible in the firmware.

garthvh, 2022-12-14:
Look at the debug log on android and you should see what happened with the request you sent, bluetooth just pulls in data that was sent over the mesh and stores it on the phone, the messaging and phone apps are meant to be 1:1 so when connecting to a bunch of different devices the local data store on the phone gets confused at some point on either android or iOS.

SR7673, 2022-12-14:
> so when connecting to a bunch of different devices the local data store on the phone gets confused at some point on either android or iOS.
Yes, that's what i observed. But it shouldn't be like that. The phone, i.e. the App knows when it is connected to a different node (different mac address) and so it should be possible to avoid that confusion. That's why i point this out here on github. I'm not one of the guys who is skilled enough to provide a solution (code) for the issue.

garthvh, 2022-12-15:
> Yes, that's what i observed. But it shouldn't be like that. The phone, i.e. the App knows when it is connected to a different node (different mac address) and so it should be possible to avoid that confusion. That's why i point this out here on github. I'm not one of the guys who is skilled enough to provide a solution (code) for the issue, sorry.

The solution is to clear the local database of messages when a new device is connected.

closed 2022-12-20 by SR7673